riscv/riscv-bitmanip #4: ANDC Bit Encoding

Keith-Cancel, 2019-04-21:
So I was looking at the encoding a lot are not fully described, but this one seems obvious too me.
`| ??????? | rs2 | rs1 | ??? | rd | 0110011 | ANDC `

Here is the current AND:
`| 0000000 | rs2 | rs1 | 111 | rd | 0110011 | AND`

Why not define ANDC much like Shift Right and Arithmetic shift right. Also ADD
to SUB is very similar. So using this bit to denote negation seems pretty intuitive. 
`| 0100000 | rs2 | rs1 | 111 | rd | 0110011 | ANDC`

While probably not as useful ANDC it's also logical easy to extend the complemented
inputs for the other bitwise instructions like this.
```
| 0100000 | rs2 | rs1 | 100 | rd | 0110011 | XORC
| 0100000 | rs2 | rs1 | 110 | rd | 0110011 | ORC
```

Also when it comes to hardware implementation negation is used when converting a positive number to negative number. So it would make easy to the same negation hardware if the same
bit was used denote negation.  Also keeping func3 the means an ALU only has worry about choosing to negate an inpute.

Keith-Cancel, 2019-04-21:
Sadly, there is no spare space for an ANDCI in the ANDI instruction. However, you usually are using bit masks that occupy a large portion of machine word. So not sure that is really much benefit.

cliffordwolf, 2019-04-21:
Are you looking at an old version of the document? Because in the current version ANDC is defined exactly like you suggest:

[image]

[image]

> it's also logical easy to extend the complemented inputs for the other bitwise instructions like this

we did and early evaluation on this over a year ago and it looks like there is very little to gain by doing this.

Keith-Cancel, 2019-04-21:
Woops, I should have looked a little more closely. The copy I have saved local has the same version number, but obviously there are differences. Should have looked at the date. 

[image]

Keith-Cancel, 2019-04-21:
Also the only time I have found the other complements would be useful is when bit slicing for constant time crypto. ANDC is handy also for muxing though.

closed 2019-04-21 by Keith-Cancel

cliffordwolf, 2019-04-21:
> [..] has the same version number [..]

"0.37-draft" just means somewhere between 0.36 and 0.37.  :)

Keith-Cancel, 2019-04-22:
I will keep that in mind.