riscv/riscv-bitmanip #5: Should be ANDN instead of ANDC.

jhauser-us, 2019-04-23:
The long-established name of the logic operation that ANDs two Boolean inputs and complements the result is NAND, not CAND.  RISC-V assembly language has a pseudo-instruction for a bitwise complement called NOT, not COMPL or whatever.  This draft extension includes instructions called NAND, NOR, and C.NOT.  For consistency, shouldn't the instruction that computes a bitwise AND with the complement of the second operand be called ANDN instead of ANDC?

Keith-Cancel, 2019-04-23:
Intel does use ANDN, but honestly neither bother me. It could also be called NORC or NORN.
Since  (¬x ∧ y) = ¬(x ∨ ¬y)

Intel BMI instructions
https://en.wikipedia.org/wiki/Bit_Manipulation_Instruction_Sets

pdonahue-ventana, 2019-04-23:
Power calls it andc:
AND with Complement
RA <- (RS) & ¬(RB)
https://openpowerfoundation.org/?resource_lib=ibm-power-isa-version-2-07-b

Alpha called it BIC (which I think was the Vax name for "bit clear"):
BIC Logical Product with Complement
Rc <- Rav AND {NOT Rbv}
https://download.majix.org/dec/alpha_arch_ref.pdf

tommythorn, 2019-04-23:
ANDN seems more consistent with NOT and current usage.

Tommy

cliffordwolf, 2019-04-26:
> The long-established name of the logic operation that ANDs two Boolean inputs and complements the result is NAND, not CAND.  [..]

I personally was aware of the PowerPC ANDC instruction for a long time, and only learned about x86 ANDN when I was doing research for the RISC-V bitmanip extension.

I think PowerPC had the ANDC instruction already in PowerPC 600 in 1993.

I think Intel added ANDN as late as in Haswell in 2013, exactly 20 years later.

So if that's right (and I'm not 100% certain), then one could argue that ANDC is the long-established name for this instruction.  :)

That being said, if anyone thinks its really important then we can change the name to ANDN.

Keith-Cancel, 2019-04-28:
I do not think it's that important. I would say again ANDC does avoid the confusion of it being some weird form of NAND.

cliffordwolf, 2019-05-28:
The latest draft now uses the name ANDN.

closed 2019-05-28 by cliffordwolf