riscv/riscv-bitmanip #6: Behavior of *w instructions

pdonahue-ventana, 2019-04-24:
The base ISA defines ADDW and SUBW like so:
> ADDW and SUBW are RV64I-only instructions that are defined analogously to ADD and SUB but operate on 32-bit values and produce signed 32-bit results. Overflows are ignored, and the low 32-bits of the result is sign-extended to 64-bits and written to the destination register.

The Bitmanip extension refers to clzw, ctzw, pcntw, etc. but doesn't actually define how they work.  The pseudocode is only defined for the instructions that operate on data of size XLEN.

Sophisticated readers understand that the *w instructions in Bitmanip are almost certainly supposed to behave like the *W instructions in the base architecture (operating on 32-bit data and then sign-extending the result to XLEN).  However, this should be explicitly addressed with some verbiage and pseudocode so that everything is fully specified and unambiguous.

pdonahue-ventana, 2019-04-24:
A related area of potential confusion is that the pseudocode for grev32 returns a uint32_t rather than int32_t.  The fact that the result is unsigned implies that the grevw instruction does not sign-extend the result to XLEN.  That would be different than the general RV64I guidance:
> These “*W” instructions ignore the upper 32 bits of their inputs and _always produce 32-bit signed values_, i.e. bits XLEN-1 through 31 are equal.

pdonahue-ventana, 2019-04-25:
Here's another related item.  Section 2.6 on Bit-matrix operations says:
> On RV128 they ignore the upper half of operands and set the upper half of the results to zero.

The second half of that sentence doesn't comply with the general philosophy that 32-bit and 64-bit operations are sign-extended.  Per RV128I version 1.7:
> The RV64I “*W” integer instructions that operate on 32-bit values in the low bits of a register are retained but now sign extend their results from bit 31 to bit 127. A new set of “*D” integer instructions are added that operate on 64-bit values held in the low bits of the 128-bit integer registers and sign extend their results from bit 63 to bit 127.

cliffordwolf, 2019-04-26:
I'd appreciate different issues for different issues.

> The base ISA defines ADDW and SUBW like so:

There is no need to add any additional language here because the base spec already states cateogically for all "*W" instruction variants in 4.2:

> Additional instruction variants are provided to manipulate 32-bit values in RV64I, indicated by a‘W’ suffix to the opcode.  These “*W” instructions ignore the upper 32 bits of their inputs and always produce 32-bit signed values, i.e.  bits XLEN-1 through 31 are equal.  They cause an illegal instruction exception in RV32I.

cliffordwolf, 2019-04-26:
> A related area of potential confusion is that the pseudocode for grev32 returns a uint32_t rather than int32_t.

It's not just grev. The reference C models for most the instructions return uint. (The only one that didn't was min/max and I've now pushed a change so that one is using uint as well.) The main reason is that signed integer arithmetic is undefined in C for things like overflow.

Also: That's not psudocode. That's actual C code that has seen extensive testing and formal verification.

cliffordwolf, 2019-04-26:
Re bit-matrix on RV128: I've now changed that to "and sign-extend the result". This way RV128 will not need *D versions of those instructions.

closed 2019-04-26 by cliffordwolf