riscv/riscv-bitmanip #7: CRC Polynomials and Encoding

Keith-Cancel, 2019-04-24:
I am not sure that hard coding the polynomials for CRC is a good idea. It seems pretty constrained to add a CRC instruction but only support 2 polynomials. Like the fact it does have 0xedb88320, what I call IEEE polynomial, and Castagnoli polynomial 0x82f63b78 does cover a large number of uses.

However, for instance here is a list of polynomials, and people may use different ones depending on the application.
https://users.ece.cmu.edu/~koopman/crc/index.html

Honestly, I think using `funct7 | rs2 | rs1 | f3 | rd | opcode | R-type` would be better than the unary format. This would allow one to load polynomials from rs2. It would also only need 2 bits in funct7 for B, H, W, and D.  Also a single bit in funct7 could select between 0xedb88320 and 0x82f63b78. There are two ways I could see handling the predefined polynomials. 

For instance if we wanted to have default to IEEE polynomial we can take advantage RS0 or the zero register by change the pseudo code to this. This also would allow anyone to load any other polynomial from a register, but it must be XOR’ed with polynomial by 0xedb88320 beforehand. For instance if I wanted to use this polynomial 0xeb31d82e XOR’ing gives me this 0x06895b0e which I can then use as the constant I load in the rs2 register.

```
uint_xlen_t crc32(uint_xlen_t rs1, uint_xlen_t rs2 int nbits) {
    for (int i = 0; i < nbits; i++) 
        rs1 = (rs1 >> 1) ^ ( ( 0xEDB88320 ^ rs2 ) & ~((rs1 & 1) - 1)); 
    return rs1; 
}
```

The other option would be to use a predefined polynomial if rs2 is the zero register. This avoids  XOR’ing the desired polynomial. Although, I am not sure XOR’ing is problem since it can easily be done once outside of run time for a given polynomial.

cliffordwolf, 2019-04-25:
> I am not sure that hard coding the polynomials for CRC is a good idea. It seems pretty constrained to add a CRC instruction but only support 2 polynomials.

Yes, I agree. That's why we also have CLMUL[H] instructions.

http://svn.clifford.at/handicraft/2018/clmulcrc/
https://www.intel.com/content/dam/www/public/us/en/documents/white-papers/fast-crc-computation-generic-polynomials-pclmulqdq-paper.pdf

I personally advocate for including CLMUL[H] in "B" and leaving the CRC instructions as optional extensions. But I think Ken Dockser would like to see it the other way around. So we will see what happens..

> This would allow one to load polynomials from rs2.

But it would significantly increase the hardware cost of the CRC unit. So you'd be better of by simply implementing the CLMUL[H] instructions and support arbitrary polynomials that way.

I'd expect the CRC instructions to be popular with small cores that do not have the hardware resource to implement an instruction like CLMUL[H] that is capable of implementing CRCs for arbitrary polynomials efficiently. So making the CRC instructions more powerful so that they support arbitrary polynomials would defeat that purpose.

closed 2019-04-25 by cliffordwolf

Keith-Cancel, 2019-04-25:
Actually, it kinda would make sense compared to all other instructions. They are generic, but crc32 is a specific algorithm instead of a generic primitive. So it makes sense as an optional extension.