riscv/riscv-bitmanip #10: Most *W instructions don't have sufficient justification

jhauser-us, 2019-05-07:
One of the frequently asked questions listed in the document is:
> Do we really need all the *W opcodes for 32 bit ops on RV64?

In my opinion, the only *W instructions in the B extension that might have significant value are the rotate instructions, RORW, ROLW, RORIW, and maybe PACKW.  To help decide, the document proposes running "proper experiments with compilers that support those instructions".  That would be ideal of course, but I'm skeptical the community will wait on the B extension long enough for that to happen.  (Plus it's not exactly easy to set up unbiased experiments for these kinds of specialized features.)

In the meantime, I'd like to point out that most of the proposed *W instructions can be substituted by a sequence of only 2 or 3 other instructions.  The following are believed to be equivalent sequences (not always unique):

**`CLZW rd,rs`**

    SLOI rd,rs,32
    CLZ rd,rd

**`CTZW rd,rs`**

    LI temp,-1
    PACK rd,temp,rs
    CTZ rd,rd

**`PCNTW rd,rs`**

    PACK rd,zero,rs
    PCNT rd,rd

**`SLOIW rd,rs1,i`**

    SLOI rd,rs1,i
    SEXT.W rd,rd

**`SLOW rd,rs1,rs2`**

    ANDI temp,rs2,31
    SLO rd,rs1,temp
    SEXT.W rd,rd

**`SROIW rd,rs1,i`**

If i > 0:

    SLLI rd,rs1,32
    SROI rd,rd,(i+32)

If i = 0:

    SEXT.W rd.rs1

**`SROW rd,rs1,rs2`**

    NOT temp,rs1
    SRLW rd,temp,rs2
    NOT rd,rd

**`GREVIW rd,rs1,i`**

    GREVI rd,rs1,i
    SEXT.W rd,rd

**`GREVW rd,rs1,rs2`**

    ANDI temp,rs2,31
    GREV rd,rs1,temp
    SEXT.W rd,rd

**`SHFLIW rd,rs1,i`**

    SHFLI rd,rs1,i
    SEXT.W rd,rd

**`SHFLW rd,rs1,rs2`**

    ANDI temp,rs2,15
    SHFL rd,rs1,temp
    SEXT.W rd,rd

**`UNSHFLIW rd,rs1,i`**

    UNSHFLI rd,rs1,i
    SEXT.W rd,rd

**`UNSHFLW rd,rs1,rs2`**

    ANDI temp,rs2,15
    UNSHFL rd,rs1,temp
    SEXT.W rd,rd

**`BEXTW rd,rs1,rs2`**

If rs2 is a known constant, rs2[63:32] = 0, and at least one bit in rs2[31:0] is a zero (very likely):

    BEXT rd,rs1,rs2

If rs2 is a known constant, rs2[63:32] != 0 (unlikely), and at least one bit in rs2[31:0] is a zero:

    PACK temp,zero,rs2
    BEXT rd,rs1,temp

If rs2 is a known constant and rs2[31:0] = 0xFFFFFFFF (unlikely):

    SEXT.W rd,rs1

If rs2 is not a known constant:

    PACK temp,zero,rs2
    BEXT rd,rs1,temp
    SEXT.W rd,rd

**`BDEPW rd,rs1,rs2`**

If rs2 is a known constant and rs2[63:31] = 0:

    BDEP rd,rs1,rs2

Else:

    BDEP rd,rs1,rs2
    SEXT.W rd,rd

**`CLMULW rd,rs1,rs2`**

    CLMUL rd,rs1,rs2
    SEXT.W rd,rd

**`FSLW rd,rs1,rs2,rs3`**

    PACK rd,rs3,rs1
    FSL rd,rd,rs2,rd
    SEXT.W rd,rd

**`FSRW rd,rs1,rs2,rs3`**

    PACK rd,rs1,rs3
    FSR rd,rd,rs2,rd
    SEXT.W rd,rd

Note that a final `SEXT.W rd,rd` can be eliminated if the rd result is known to be used only in subsequent 32-bit operations (such as SW or other *W instructions).  Other optimizing tweaks are also possible, depending on the circumstance.

Unless the proposed *W instructions can be shown to be much more prevelant than I expect, the combination of rare utility plus relatively easy synthesis from other instructions argues strongly for dropping them.

The document also says:
> But they add very little complexity to the core. So the only question is if it is worth the encoding space.

While "very little complexity" may be true, I disagree that it should be dismissed and only encoding space considered.  "Very little complexity" certainly lowers the threshold of utility an instruction must demonstrate to be acceptable, but it doesn't make the instruction free to add.  There are many other possible instructions of very little complexity that we so far choose to exclude, and these *W instructions perhaps should be among them.

For instance, there's a whole category of instructions that would have more impact but aren't currently included, and that is the unsigned equivalents of the existing RV64I *W instructions:  ADDWU, SUBWU, SLLWU, etc.  These would be just like the existing *W instructions but instead zeroing the upper 32 bits, as appropriate for an `unsigned int` or `uint32_t` result type rather than `int` or `int32_t`.  We hardly need to run any experiments to know that such *WU instructions would be used far more frequently than the *W instructions proposed for the B extension.

cliffordwolf, 2019-05-08:
Hi John.

>To help decide, the document proposes running "proper experiments with compilers that support those instructions". That would be ideal of course, but I'm skeptical the community will wait on the B extension long enough for that to happen.

The community will need to wait for this anyway because this determines other decisions about what to include in the B extension as well.

On the contrary, to rush out a final B-extension spec before we even have compiler support and evaluation results would be very reckless.

> In the meantime, I'd like to point out that most of the proposed *W instructions can be substituted by a sequence of only 2 or 3 other instructions.

This is the case for most *W instructions and is completely missing the point of *W instructions. 

If you have 32-bit code it would be completely unacceptable if it would run 2-3x faster on RV32 than on RV64. The *W instructions must be seen as a whole under this objective, not as individual instructions.

Another aspects of *W is that multi-cycle operations could be faster if the core already knows its only a 32-bit operation. Examples would be MULW, BEXTW, BDEPW.

In many of the cases you have listed, the extra effort in the instruction decoder to blacklist those *W instructions in the instruction decoder would be larger than the extra effort to implement the *W instructions.

For example, the ALU hardware cost for having FSLW/FSRW when you already have FSL/FSR/ROLW/RORW is one extra ANDNOT gate. https://github.com/riscv/riscv-bitmanip/blob/master/verilog/shifter64.v#L51 Obviously the decoder cost for blacklisting it would be larger.

Finally, the *W instructions can also be used to reduce power by clock-gating half of the data path. This would even work with non-W instructions as long as they have the invariant that 32-bit sign-extended inputs will produce 32-bit sign-extended outputs. For this, one would simply tag all 32-bit sign-extended values in the register file. But that requires that all instructions that do not have this invariant have a *W variant. That was the main criteria used to determine which instructions should have a *W variant. The only exception to that rule is that there is no CLMULHW, like there's also no MULHW.

> While "very little complexity" may be true, I disagree that it should be dismissed and only encoding space considered. "Very little complexity" certainly lowers the threshold of utility an instruction must demonstrate to be acceptable, but it doesn't make the instruction free to add.

As I have said, in many of the cases you have mentioned it would even require more hardware to blacklist those instructions.

I mean look at your own code:

> ANDI temp,rs2,15
> SHFL rd,rs1,temp
> SEXT.W rd,rd

The overhead for SHFL/SHFLW is ONE SINGLE EXTRA GATE!  The only thing required is masking the upper bit of the argument. (Sign-extending the result of *W operations can be done based on the major opcode alone. No per-instruction overhead here.)

Clearly the extra effort to blacklist this instruction in the decoder would be larger than one single extra gate!

And for SHFLIW:

SHFLI rd,rs1,i
SEXT.W rd,rd

That's just free. It costs you nothing. If you have any *W operations then you already have the MUXes for the sign-extend, driven by the major opcode. All that blacklisting SHFLIW would do is driving up the cost and complexity of the instruction decoder.

> Unless the proposed *W instructions can be shown to be much more prevelant than I expect, the combination of rare utility plus relatively easy synthesis from other instructions argues strongly for dropping them.

A few paragraphs above you dismissed the idea of such an evaluation because as you said "I'm skeptical the community will wait on the B extension long enough for that to happen".

We need to implement those things in hardware and compilers and then measure. This is the current roadmap. You can't say both, that those instructions must prove themselves in such an evaluation, and that we don't have time to conduct that evaluation.

> While "very little complexity" may be true, I disagree that it should be dismissed and only encoding space considered.

Yet you are arguing against collecting more data and for dropping them now based on your suspicions on how commonly they are used and what the hardware cost of those instructions would be? That's not a very convincing argument.

> For instance, there's a whole category of instructions that would have more impact but aren't currently included, and that is the unsigned equivalents of the existing RV64I *W instructions: ADDWU, SUBWU, SLLWU, etc.

Now you are just silly. You are proposing to spend TWO MAJOR OPCODES on nothing else than unsigned versions of the *W instructions? You realize that this would be 66% of the remaining reserved major opcodes, yes?

jhauser-us, 2019-05-08:
> If you have 32-bit code it would be completely unacceptable if it would run 2-3x faster on RV32 than on RV64.

I'm not certain what you mean by "32-bit code" here.  RV32 code won't run directly on an RV64 machine.  You may be imagining code that uses 32-bit integer types and 32-bit "intrinsics", such as `__builtin_shfl`, as opposed to the 64-bit "long" version, presumably `__builtin_shfll`.  The glitch in that argument is that 32-bit `unsigned int` can be as likely to appear as `int`, and then the *W instructions aren't the right ones, because they sign-extend instead of zero-extend.  So you end up with extra instructions to account for the difference, and the 32-bit code again runs slower---in the worst case, maybe 2x slower---on RV64 than on RV32, even though we called that "completely unacceptable".

Ideally, 64-bit machines would be running code written for them, as that may even double the performance for some of these bit-manipulation applications.  I'm personally more interested in that case than the question of how fast 32-bit code runs on RV64.

> In many of the cases you have listed, the extra effort in the instruction decoder to blacklist those *W instructions in the instruction decoder would be larger than the extra effort to implement the *W instructions.  

That's not as definite as you make it sound.  It depends on the implementation.

> (Sign-extending the result of *W operations can be done based on the major opcode alone. No per-instruction overhead here.)

Only when the implementations of the B instructions in question share part of their pipeline path with other *W instructions.  You may be assuming that would always be true, but it isn't necessarily.

> And for SHFLIW:
>
> SHFLI rd,rs1,i
> SEXT.W rd,rd
>
> That's just free. It costs you nothing. If you have any *W operations then you already have the MUXes for the sign-extend, driven by the major opcode.  

Again, you're assuming that the implementations of those different instructions converge in the pipeline.  For many implementations, that will happen naturally, but it isn't necessarily true.

> A few paragraphs above you dismissed the idea of such an evaluation because as you said "I'm skeptical the community will wait on the B extension long enough for that to happen".
>
> We need to [...]

Please correct me if I'm wrong, but I can't see how compiling generic, standard C code is going to produce many of these bit-manipulation instructions automatically, where the compiler matches source code patterns to produce BEXT, SHFL, etc.  Instead, it seems to me, the sort of source code that will use these instructions will depend heavily on intrinsics, like `__builtin_clz`, `__builtin_popcount`, and, presumably, `__builtin_bext`, etc.  In that environment, where the intrinsics and the machine instructions are more-or-less one-to-one, the usage numbers you get will be strongly biased by the source code examples you choose.

If you want to "prove" that BEXT is the most important instruction, just make sure all your code examples use `__builtin_bext` frequently.  You can then measure the frequency of use, and pretend it's all very scientific, but those numbers will be highly tainted by the original selection of the input samples.

What I meant was, I expect that figuring out some way to perform a set of reasonably *unbiased* experiments would take several years, if it's even possible at all, and, no, I don't think the community really wants to wait that long.  If you have some better way to gather data than what I described, I'd be happy to hear of it.

> Now you are just silly. You are proposing to spend TWO MAJOR OPCODES on nothing else than unsigned versions of the *W instructions? You realize that this would be 66% of the remaining reserved major [...]

First off, I was only pointing out that they would be more useful, and more often used, than the *W instructions you feel are so necessary.

Secondly, you are incorrect.  There is existing room in the OP-IMM-32 and OP-32 groups for encoding such *WU instructions, if a decision were made to adopt them.  No additional major opcodes need to be allocated.

I'm willing to accept pushback and engage in debate on these matters, but I was disappointed that you closed this issue so swiftly, before anyone else might comment.  I think it would be better, after providing your reasons for rejecting, if you left issues open for a little longer, so others might see and have an opportunity to correct anything that might need correcting.

cliffordwolf, 2019-05-08:
> I'm not certain what you mean by "32-bit code" here.

For example code using `uint32_t`. That's literally the whole motivation for having the *W instructions. So I would have assumed it is clear from the context.

And I feel like you simply disagree with the general RISC-V design decision to have *W instructions in the first place, but the B extension is hardly the right place to "fix" this decision.

> The glitch in that argument is that 32-bit unsigned int can be as likely to appear as int, and then the *W instructions aren't the right ones, because they sign-extend instead of zero-extend. So you [...]

That's just plain wrong.

The idea behind *W instructions is simply that they operate on the lower 32 bits. It makes sense to have a consistent scheme for how to fill the upper bits, but it doesn't matter much what this scheme is exactly, if that scheme is sign-extend or zero-extend.

In fact, the code most benefiting from *W is code using `uint32_t`, not code using `int32_t`, because in C a signed int overflow is undefined but an unsigned overflow is defined. So you can just use 64-bit ops on an `int32_t` but for `uint32_t` you need *W instructions.

> That's not as definite as you make it sound. It depends on the implementation.

Well, yes, but if you follow this argument all the way to the end you'd have to conclude we can't know anything and therefore can't make an informed decision either way. But that doesn't sound very productive.

So why not instead make some sane assumptions and look at the obvious ways of implementing this across a range of differently sized cores. Or implement it and measure. We are going to do the latter one and afaict what I said is broadly true for the former one.

> Please correct me if I'm wrong, but I can't see how compiling generic, standard C code is going to produce many of these bit-manipulation instructions automatically

That depends on the instruction. Rotate shifts for example are something I'd absolutely expect a C compiler to infer. Same with SLO/SRO. Funnel shifts as well.

I don't expect a C compiler to compile generic code into GREV/SHFL/UNSFL but there are code patterns that match well to instruction sequences utilizing GREVI/SHFLI/UNSHFLI, especially BSWAP (GREVI -8) and BREV (GREV -1). Most compiler can already generate code using BSWAP.

There are things that could theoretically map well to certain sequences using BEXT/BDEP but most compiler's don't use them yet. But Intel and AMD only added their BEXT/BDEP equivalent a few years ago, so just give compilers a little more time to catch up.

> Again, you're assuming that the implementations of those different instructions converge in the pipeline. For many implementations, that will happen naturally, but it isn't necessarily true.

But it makes sense to optimize for the case we are expecting, not for the case we are not expecting.

If removing functionality makes the implementations we expect more complex, I don't see how it makes sense to remove that functionality. Especially if it's only a few extra gates for the implementations we don't expect.

>  Instead, it seems to me, the sort of source code that will use these instructions will depend heavily on intrinsics, like __builtin_clz, __builtin_popcount

Note that `__builtin_clz` and `__builtin_popcount` both operate on 32-bit integers when using the RV64 calling convention. So they map directly to CLZW, PCNTW.

You'd have to use `__builtin_clzl` and `__builtin_popcountl` for the 64-bit version.

So all the existing code using `__builtin_clz` and `__builtin_popcount` will be sped up by the existence of CLZW and PCNTW.

> and, presumably, __builtin_bext, etc.

Currently GCC has intrinsics `__builtin_ia32_pext_si`, `__builtin_ia32_pext_di`, `__builtin_ia32_pdep_si`, and `__builtin_ia32_pdep_di`. `<x86intrin.h>` provides wrappers `_pext_u32`, `_pext_u64`, `_pdep_u32`, and `_pdep_u64`.

Yes, I think it would sense to introduce architecture-independent intrinsics for these operations. I would assume this will happen when RISC-V also has instructions with this semantic.

> If you want to "prove" that BEXT is the most important instruction, just make sure all your code examples use __builtin_bext frequently. You can then measure the frequency of use, and pretend it's [...]

Obviously we wouldn't measure the frequency of use as even a high frequency of use wouldn't tell us if the instruction is any good.

The relevant metrics are runtime speedup and code size reduction. Even if you hand-optimize an example to use something like BEXT/BDEP. If that leads to runtime speedup and/or code size reduction then this demonstrates the usefulness of the instruction in this context.

As with any microbenchmark this leaves the question open how relevant the example is for a larger code base. That's why we also need to do things like measure CRC/CLMUL instructions "in vivo" for stuff like the decompression and CRC checking code used in the Linux kernel startup.

> First off, I was only pointing out that they would be more useful, and more often used, than the *W instructions you feel are so necessary.

I think you still haven't understood what the *W instructions are actually for.

They are for operating on the lower 32-bit of your 64-bit values. How you pad the results to 64-bit is not relevant for the intended use case.

[... cut ...]
So the only space "left" in OP-IMM-32 and OP-32 are opcodes that correspond to OP-IMM/OP instructions that do not have a *W variant. And that's not a lot of free space.

But again. This is a conversation you should be having with Andrew. and it doesn't seem to have anything to do with the B extension. You seem seem to not like having *W instructions, and/or prefer *WU instructions. There is nothing the B extension could do about that.

> I'm willing to accept pushback and engage in debate on these matters, but I was disappointed that you closed this issue so swiftly, before anyone else might comment.

I did re-open the issue.

brucehoult, 2019-05-08:
I agree with @cliffordwolf. If you use an LP64 model then you need to make sure all the code out there that uses int or unsigned int runs well on a 64 bit CPU.

As pointed out, functions such as __builtin_popcount() take unsigned int, not unsigned long int.

If you provide *W operations you have to decide what to do with the upper bits. You can leave then alone (x86), zero extend them (Aarch64), or sign extend them (RISC-V). If you leave them alone then casting a 32 bit value to a 64 bit value requires a sext or zext every time. If you sign extend them then only unsigned values require a zext, signed ones are already correct. If you zero extend them then only signed values require a sext, unsigned ones are already correct. It's hard to say which is better. Most normal application code uses signed more than unsigned, favouring sign extension, and this also means you don't need special 32 bit compare (or compare and branch) instructions as the 64 bit instructions produce the correct result for both signed and unsigned 32 bit values.

It's a bit unfortunate that Aarch64 and RISC-V picked opposite answers. If they weren't being designed effectively at the same time (Aarch64 was designed first, but in secret, and announced after RISC-V) that might not have happened. It's not terribly important except in micro-benchmarks, where it's pretty much luck of the draw (or design) which one is favoured.

It's also a bit unfortunate that in the base instruction set RISC-V needs two instructions to zero-extend. These are an obvious choice for a macro-op fusion pair, and also the proposed PACK instruction can be used as a zero-extend.

Needing an extra instruction when unsigned is cast to unsigned long is unfortunate. Needing one or two extra instructions any time anything at all is done with 32 bit variables is, yes, unacceptable.

pdonahue-ventana, 2019-05-08:
The decision should be framed by the criteria mentioned in section 1.1 of the Bitmanip draft.

Architecture consistency: Not having *W would be inconsistent with the architecture.  As far as I can tell, every instruction in the base architecture has a W equivalent (either as a new W instruction or, assuming all other *W instructions sign-extend the result, using an existing instruction).

Threshold metric: It depends on the exact threshold but at least some of the B extension's proposed W instructions don't meet the heuristic threshold.  But many W instructions in the base architecture also don't meet this metric (or the architecture's "small base integer ISA" goal).  Most of them can use the 64-bit instruction and then SEXT.W.  The decision was made that having the *W instructions across the board (architectural consistency) is more important than having fewer instructions.

Data-driven value: The FAQ says that *W will be taken out if there is no data.  The data-driven value criterion suggests that they shouldn't be added until there is data.  Or maybe the data-driven value criterion only applies to having the basic instruction plus the W variant, not each independently.  If that's the case then each instruction has already met the data-driven value test.

Hardware simplicity: It seems relatively simple either way.  Requiring an extra gate to remove *W doesn't seem that important but requiring several extra gates on some implementation that has a whole separate datapath (which itself adds far more gates) also doesn't seem that important.

Compiler support: This would seem to be either neutral or favor having *W.

For consistency, I think that it makes sense to keep the *W instructions regardless of whether there's data that they each individually have merit.  (Unless, of course, any of them is equivalent to a single other instruction.  That's why there's already no andcw or minw.)

brucehoult, 2019-05-08:
"Data-driven value: The FAQ says that *W will be taken out if there is no data. The data-driven value criterion suggests that they shouldn't be added until there is data."

If you don't include something in the set of what you'll gather data on, then you'll never have the data to decide whether it should be in the set of things in the eventual standard extension.

I agree that *W versions don't have to justify themselves. We do, as a matter of consistency, have *W instructions. The 32 bit and 64 bit version should be evaluated together.

jhauser-us, 2019-05-09:
I'd like to thank everyone who has commented.

I wrote:
> The glitch in that argument is that 32-bit unsigned int can be as likely to appear as int, and then the *W instructions aren't the right ones, because they sign-extend instead of zero-extend. So you [...]

Someone has since pointed me to a comment in the ISA specification's chapter for base RV64I:
> The compiler and calling convention maintain an invariant that all 32-bit values are held in a sign-extended format in 64-bit registers.  Even 32-bit unsigned integers extend bit 31 into bits 63 through 32.

I saw this many months ago and since forgot about it, since I've been working directly with RV32 most of the time.  Based on past experience, I automatically expected that signed 32-bit values needed to be sign-extended to 64 bits and unsigned 32-bit values zero-extended to 64 bits, so that the 64-bit values in the registers would match the intended 32-bit values.  Under that mistaken assumption, the *W instructions wouldn't do the right thing for `unsigned int` and `uint32_t`.  Since the unsigned types could reasonably be expected to predominate over signed `int` and `int32_t` for the data that B instructions operate on, I couldn't see a good justification for providing hardware specially for the sign-extension case only.

Now that I've been corrected, it's easier for me to accept the *W versions of the B instructions.  You're right that I'm not entirely comfortable with the convention of sign-extending unsigned values, but I understand that the choice is set in stone now, so we have to work with it.  Sorry that I missed this earlier.

Clifford Wolf wrote:
> I think you still haven't understood what the *W instructions are actually for.
> 
> They are for operating on the lower 32-bit of your 64-bit values. How you pad the results to 64-bit is not relevant for the intended use case.

While that's an extremely important use case, it's not the only one.  It is common for ordinary 64-bit code to have a mix of 64-bit and 32-bit types, such as `long` and `int`, with 32-bit `int` values commonly participating in 64-bit `long` or pointer computations.  In such cases, it very much matters how the upper 32 bits get filled, whether with zeros or an extension of the sign.  RISC-V's choice of sign-extension for *W was made deliberately to give preferential support for signed integers, so it's not quite right to say  the choice was arbitrary.

The part I was missing yesterday was the ABI rules that require sign extension of *unsigned* 32-bit values.  That convention is bad if you want  to use those values in 64-bit arithmetic, but it's good for standardizing on the existing *W instructions.

> So the only space "left" in OP-IMM-32 and OP-32 are opcodes that correspond to OP-IMM/OP instructions that do not have a *W variant. And that's not a lot of free space.

You might want to check again.  Excluding SLT, SLTU, XOR, OR, and AND leaves more than half the space open.

> I did re-open the issue.

And I appreciate it.

cliffordwolf, 2019-05-09:
> You might want to check again. Excluding SLT, SLTU, XOR, OR, and AND leaves more than half the space open.

That may help in OP-IMM-32 that's already very full and where is not a lot of space for new instructions in OP-IMM.

But, and maybe I misunderstood what you were proposing, there is no guarantee that in the future less than half of the OP instructions will have a *W variant. In fact, look at the M extension for example. It has 8 OP instructions and 5 OP-32 instructions. If you wanted to add *WU instructions for M you would need to use one of the "holes" the base ISA leaves. So if you want to add a *WU instruction for every *W instruction it doesn't matter what percentage of the _base_ ISA instructions leave "holes" in OP-32. The only thing that matters what percentage of _all_ OP instructions, including future ones, will have *W variants.

So what I'm saying is, maybe there's enough concrete free space for *WU instructions in OP-IMM-32 and OP-32 for *WU variants of _base_ ISA instructions. But there surely isn't enough _systematic_ space to add *WU variants of all OP instructions.

Also, as I'v said before, adding *WU instructions in the holes in OP-32 would make a huge mess in any instruction decoder.

jhauser-us, 2019-05-10:
> So what I'm saying is, maybe there's enough concrete free space for *WU instructions in OP-IMM-32 and OP-32 for *WU variants of base ISA instructions. But there surely isn't enough systematic space to add *WU variants of all OP instructions.

I agree it would be terrible to do such a thing ad hoc, instruction by instruction.  If the Foundation chooses to, one option is to adopt a system like the following that I believe is compatible with all current standard instructions:

    instruction bits
    25 14 13  6:2

     -  -  0  00100   group A, imm     (ADDI, SLLI, SRLI, ...)
     -  -  1  00100   group B, imm     (SLTI, SLTIU, ORI, ANDI)

     -  -  0  00110   group A, imm, W  (ADDIW, SLLIW, SRLIW, ...)
     -  -  1  00110   group A, imm, WU (ADDIWU, SLLIWU, SRLIWU, ...)

     0  -  0  01100   group A, rs2     (ADD, SLL, SRL, ...)
     0  -  1  01100   group B, rs2     (SLT, SLTU, OR, AND, ...)
     1  0  0  01100   group C, rs2     (MUL, MULH, ...)
     1  0  1  01100   group D, rs2     (MULHSU, MULHU, ...)
     1  1  -  01100   group E, rs2     (DIV, DIVU, ...)

     0  -  0  01110   group A, rs2, W  (ADDW, SLLW, SRLW, ...)
     0  -  1  01110   group A, rs2, WU (ADDWU, SLLWU, SRLWU, ...)
     1  0  0  01110   group C, rs2, W  (MULW, ...)
     1  0  1  01110   group C, rs2, WU (MULWU, ...)
     1  1  -  01110   group E, rs2, W  (DIVW, DIVUW, ...)

The group labels A through E are simply my invention.  As you can see, instructions in groups A and C have encoding space for 32-bit versions, both for sign extension (W) and zero extension (WU).  Instructions in groups B and D have no 32-bit versions, and those in group E can have W versions but not WU versions.  It's not perfect, but any system must navigate around the existing instructions.

To be clear, there would be no change to the existing RV64I base.  Any added *WU instructions would be part of an extension.

Andrew has told me he wants to keep open the option for *WU instructions for now, which implies to me we should devise a system to reserve the encoding space now, even if the idea eventually gets dropped.  I haven't run this particular system by him yet, but I intend to do so soon.  

Naturally, we would want any system that's adopted to be fully compatible with the B extension, by tweaking either or both as necessary.  I'll be looking into this question soon.  And now you or anyone else can do so too, if you're so inclined.

But this is getting off-topic for this GitHub issue, so we should move the discussion elsewhere if you'd like to continue.

brucehoult, 2019-05-10:
I agree that there is merit in systematically adding *WU versions of R-type instructions that have *W versions. If you do it at all then it should be for BOTH the base instruction set and for any new *W instructions in the BitManip extension.

This is easily affordable in terms of opcode space by, as has been pointed out, using something in the hi bits of the instruction, keeping the identical opcode and func3 to the *W version. It's also very cheap to implement.

Probably the only OP-IMM-32 instruction that can be justified is ADDIWU. That *does* need a new opcode.

This can't be retrofitted to the base instruction set. It needs to be in an extension. There is a real and multiplying cost to every extra extension.

I know Clifford has said this doesn't fit into the BitManip extension. However, since we'd like to have some *WU instructions in the BitManip extension then we should *at minimum* make sure that the encoding scheme used is compatible with another extension adding *WU versions of the base instruction set *W instructions.

However I think adding *WU to base ISA *W instructions would not be a ridiculous thing to do in the BitManip extension. It could save proliferation of extensions, it's very low cost (and high benefit), it could potentially encourage people to implement the BitManip extension who might not otherwise do so [1]. If evidence was gathered showing it to be useful (pretty trivial -- it would eliminate virtually every "LSL n;LSR n" sequence in RISC-V code) then I think Andrew could be persuaded to put it in B.

[1] it could also be separately available in another extension for people who really really don't want to implement shuffles etc -- multiple extensions are allowed to implement the same instruction.

jhauser-us, 2019-05-10:
Interestingly, in a message to me yesterday where we talked about possible *WU instructions, Andrew literally wrote:
> I’m not opposed to putting them in B.

To be sure, that expresses ambivalence more than advocacy.  But as far as "could be persuaded" goes, I believe yes.

cliffordwolf, 2019-05-10:
> This is easily affordable in terms of opcode space by, as has been pointed out, using something in the hi bits of the instruction

So that would set 50% of the OP major opcode aside for stuff that has no *W equivalent?

Also, in case there will be more ternary instructions in the future: I don't see that scheme working for ternary instructions, where there is only encoding space for two ternary instructions per minor opcode.

> Probably the only OP-IMM-32 instruction that can be justified is ADDIWU

One would also want an SLTIWU instruction, as this is not equivalent to SLTI.

The shifts don't need that much encoding space.

So, if for example `inst[31]` is used to indicate *WU instructions in OP-32 and OP-IMM-32 (shifts only) then the only things required additionally would be

- Two major opcodes in OP-IMM-32 for ADDIWU and SLTIWU
- Ternaries would need special handling.

For B we could put FSLWU/FSRWU in the places for the non-existing CMIXW/CMUXW instructions. I would not know where to put FSLIWU/FSRIWU.

> I know Clifford has said this doesn't fit into the BitManip extension.

I can be persuaded. :) Especially when Andrew says he is not against it. I would have expected more pushback for something that uses this much encoding space.

So if enough people in the group want this then I'd be happy to write it up. Sounds like the kind of thing that should be easy enough to evaluate.

I personally am a bit concerned about the encoding space that this would require (half of OP-32 + two minor ops in OP-IMM-32, plus effectively 50% of OP). And it would always just replace two instructions: a *W instruction followed by a PACK to zero-out the upper half. Considering that this is only useful when converting from 32-bit values to 64-bit values, I'm not sure how many instructions it would safe in real-world code. (I think code size is the more interesting metric here then performance considering that a high-performance processor could easily merge *W+PACK.)

For comparison: A load-immediate instruction that can load a 17-bit signed immediate would only require a single minor opcode and would replace a 48-bit sequence with a 32-bit sequence. Tbh I would not be surprised if such an instruction would reduce the code size more than adding *WU instructions, at a fraction of the encoding space cost. (But I'm just speculating here of course.)

cliffordwolf, 2019-05-10:
I wrote:
> I would not know where to put FSLIWU/FSRIWU.

Actually, since there would only be FSRIWU it would fit nicely where the non-existing FSLIW instruction would go.

I wrote:
> Ternaries would need special handling.  
> For B we could put FSLWU/FSRWU in the places for the non-existing CMIXW/CMUXW instructions.

Similarly, half of the OP ternary encodings, i.e. half of the opcodes, could be reserved for ternary OPs that don't need *W/*WU variants, so that we could systematically use those encodings in OP-32 for *WU.

Bruce wrote:
> Probably the only OP-IMM-32 instruction that can be justified is ADDIWU

What I really like about ADDIWU is that it would enable a two-instruction (64-bit) sequence for loading an unsigned 32-bit immediate into a 64-bit register. Right now this needs three instructions.

cliffordwolf, 2019-05-10:
I wrote:
> So that would set 50% of the OP major opcode aside for stuff that has no *W equivalent?

I haven't seen the P extension proposal yet but I'd assume they would want a lot of encoding space, and would not need *W variants of their instructions. So that would be one example of something that could go in that space (presumably with lots of space to spare, half a major opcode is huge).

cliffordwolf, 2019-05-12:
Moved *WU discussion to #14. Closing here.

closed 2019-05-12 by cliffordwolf