riscv/riscv-bitmanip #11: Bring sanity to source operand order for ternary instructions

jhauser-us, 2019-05-07:
Please, please, please, in the assembly language for the ternary instructions, do not place the control operand of CMIX between the other two source operands, do not place the condition operand of CMOV between the other two source operands, and do not place the shift amount for funnel shifts (FSL and FSR) between the other two source operands.

I understand the hardware motivation for having the control and shift amounts be in rs2, which forces the other two operands to be rs1 and rs3.  But it would be better to define the assembly language for these instructions as

    CMIX rd,rs2,rs1,rs3
    CMOV rd,rs2,rs1,rs3
    FSL  rd,rs1,rs3,rs2
    FSR  rd,rs1,rs3,rs2

Whatever extra trouble a nonlinear operand order might cause for tools authors, it is nothing compared to the multiplicative effect of foisting an illogical order on programmers.  Let's not forget, there are literally thousands of programmers for every tools author, and we'd prefer as often as possible to help those programmers write bug-free code.

I note that assembly language pseudo-instructions already provide some precedent for breaking a definite connection between operand order and source register numbers.  Store instructions are another existing exceptional case, being written as

    SW rs2,offset(rs1)

and not

    SW rs1,rs2,offset

asb, 2019-05-07:
FWIW, having a different order in the assembly isn't a big deal for LLVM at least. So the cost for tool authors is very very low.

I'm strongly in favour of picking whatever operator order is most logical for programmers.

cliffordwolf, 2019-05-08:
Sounds good to me. I'm not sure if

```
CMIX rd,rs2,rs1,rs3
CMOV rd,rs2,rs1,rs3
```

would be better or

```
CMIX rd,rs1,rs3,rs2
CMOV rd,rs1,rs3,rs2
```

I think "control argument last" would be a bit more intuitive.

cliffordwolf, 2019-05-09:
People on the task group mailing list liked your cmix/cmov argument ordering (rs2 first) more than mine (rs2 last). So rs2 first it is. I updated the document accordingly.

closed 2019-05-09 by cliffordwolf