riscv/riscv-bitmanip #9: documentation issue

eroom1966, 2019-05-07:
@cliffordwolf 
Hi Clifford, 
I have a query regarding the documented behavior for fsl, the text says that 
The fsl rd, rs1, rs2, rs3 instruction creates a 2 x XLEN word by concatenating rs1 and rs3
(with rs1 in the MSB half)
from the pseudo code, it looks as though rs1 is in the lower half.
can you clarify which is correct ?
Thx
Lee

cliffordwolf, 2019-05-07:
I think I know what the misunderstanding is: FSL returns the _MSB_ half of the shift result, not the LSB half. If you'd define FSL as returning the _LSB_ half of the shift result then the halfs would be swapped. Quoting the FSL spec:

> [..] and then writes the MSB half of the result to rd.

eroom1966, 2019-05-07:
aha, RTFM (more than once anyway) thanks
I will close this issue

closed 2019-05-07 by eroom1966