riscv/riscv-bitmanip #17: Suggest renaming GREVI pseduo-instructions to mirror ZIP/UNZIP

jhauser-us, 2019-05-17:
The pseudo-instructions defined for GREVI (BREV.P, PSWAP.N, etc.) are like the ZIP and UNZIP pseudo-instructions in that they move "units" of a power-of-two size within "components" of a larger power-of-two size.  For the ZIP/UNZIP pseudo-instructions, there is a simple pattern of

    ZIP<unit-size><component-suffix>
    UNZIP<unit-size><component-suffix>

In this system, &lt;unit-size&gt; is either empty, meaning 1 bit, or is a decimal number of bits ("2", "4", "8", or "16"); and &lt;component-suffix&gt; is either empty, meaning the full register size (XLEN), or is one of the suffixes '.N', '.B', '.H', or '.W'.

It would aid comprehension if the GREVI pseudo-instructions followed the same system.  I propose 

    REV<unit-size><component-suffix>

This would rename all of the GREVI pseudo-instructions as follows:

    BREV.P  -> REV.P
    PSWAP.N -> REV2.N
    BREV.N  -> REV.N
    NSWAP.B -> REV4.B
    PSWAP.B -> REV2.B
    BREV.B  -> REV.B
    BSWAP.H -> REV8.H
    NSWAP.H -> REV4.H
    PSWAP.H -> REV2.H
    BREV.H  -> REV.H
    HSWAP.W -> REV16.W
    BSWAP.W -> REV8.W
    NSWAP.W -> REV4.W
    PSWAP.W -> REV2.W
    BREV.W  -> REV.W
    WSWAP   -> REV32
    HSWAP   -> REV16
    BSWAP   -> REV8
    NSWAP   -> REV4
    PSWAP   -> REV2
    BREV    -> REV

If the name "BSWAP" is so entrenched that we feel we must have this mnemonic, then BSWAP can be made another pseudo-instruction alias for REV8.

cliffordwolf, 2019-05-18:
> It would aid comprehension if the GREVI pseudo-instructions followed the same system.

That must be the first time some says anything along the lines of "if only GREV would be as easy to understand as ZIP". Usually it's the other ways around..  :)

> If the name "BSWAP" is so entrenched that we feel we must have this mnemonic, then BSWAP can be made another pseudo-instruction alias for REV8.

That's a very complicated thing for me to assess, because in the end naming things is about aesthetics, and I'm already so used the the existing naming scheme that everything new feels alien..

I'll raise the issue on the mailing list.

> REV\<unit-size\>\<component-suffix\>

I personally would read REV8 as "Revision 8", which I would find a bit confusing.

But I'm not opposed to to the \<unit-size\>\<component-suffix\> approach.

However, tbh, I'm not even sure if there should be that many pseudo ops for all the different GREV/SHFL/UNSHFL control values. Originally I've just added them because I though it might make it easier to understand what kind of operations those instructions could perform.

The most often used ones are definitely BSWAP (-8), BREV (-1), ZIP (-1), and UNZIP (-1), followed by BREV.P (1) and ZIP.N (1). Maybe a better solution would be to find good descriptive names for BREV.P and ZIP.N and not bother with other pseudo-ops? Idk..

The spec document would certainly be improved if it contained some information on how to come up with the GREVI/SHFLI/UNSHFLI control values, so nobody needs to look them up in a table.

I mean on RV32, `GREVI rd, rs, 16` is equivalent to `RORI rd, rs, 16`. The former has a special name, `HSWAP`, but giving a name to the latter would be a bit ridiculous. And I think the reason is that most programmers have a very good understanding of what `RORI rd, rs, 16` does on a 32-bit machine, but in `GREVI rd, rs, 16` the `16` is just a magic number.

> or is one of the suffixes '.N', '.B', '.H', or '.W'.

Or '.P' or course. :) This suffix doesn't exist for shuffle because you need at least four elements for a shuffle, but reversal works with two elements as well.

cliffordwolf, 2019-06-04:
I've changed the grevi pseudo-instruction naming scheme to revN.S in the latest draft spec.

closed 2019-06-04 by cliffordwolf