riscv/riscv-bitmanip #18: Rename ADDUW, SUBUW, SLLIUW

jhauser-us, 2019-05-17:
The new "prefix zero-extend" instructions currently have names that don't follow the existing convention for *W instructions.  In particular, these new instructions do not act at all like instructions DIVUW and REMUW, which perform their operation on two unsigned 32-bit values and then sign-extend the 32-bit result.

To avoid confusion, the new instructions need different names.  I propose

    ADDZX
    SUBZX
    SLLZXI

where "ZX" stands for "zero-extend".  For example, instruction

    SLLZXI rd,rs1,i

acts the same as the sequence

    ZEXT.W rd,rs1
    SLLI rd,rd,i

pdonahue-ventana, 2019-05-17:
ADDZX doesn't convey that it's a 32-bit add which is zero-extended.  This could cause confusion on RV128 where there would presumably be a 32-bit zero-extended add and a 64-bit zero-extended add (much like there are ADDW and ADDD for sign-extended adds).  How about ADDWZX, SUBWZX, SLLWZXI which allows for ADDDZX, SUBDZX, SLLDZXI on RV128?

jhauser-us, 2019-05-17:
> How about ADDWZX, SUBWZX, SLLWZXI

Okay by me.  (In some respects it's imperfect, but I can't think of a better arrangement.)

cliffordwolf, 2019-05-18:
The original name would have been `ADDWU_RS2`, so `ADDUW` is already is an improvement.  ;)

> ADDZX

As @pdonahue-ventana pointed out, we need the W/D in there so that it's clear is if't zero-extending a 32-bit or 64-bit input.

> have names that don't follow the existing convention for *W instructions

Since they do not have the semantic of *W instructions I don't understand in what way that would matter.

> ADDWZX, SUBWZX

I think there should be a consistent naming scheme for what's currently called *WU and *UW instructions. I'm fine with renaming it *ZX or whatever, but only as part of a proposal that also renames the *WU instructions.

The name should include the following information:
- The size of the word we are zero-extending (W/D)
- If the zero-extension happens post- or prefix (rs1 for SLLIUW, rs2 for ADDUW/SUBUW)

> SLLWZXI

If at all then SLLIWZX.

jhauser-us, 2019-05-24:
For the *WU names, I would say there is existing precedent in the load instructions:  LWU acts the same as LW, except for zero-extending the 32-bit result, and likewise, LHU and LBU act the same as LH and LB except for zero-extending the result for halfword and byte sizes.  For this reason, I find it consistent and unsurprising that ADDWU would act the same as ADDW except for zero-extending the word-size result.

Yes, it would be nice if we could have a complete naming system that would encode all the variances we want in a logical and succinct way, but I don't think that's going to be possible.  The best we can do, I think, is to try to avoid unnecessary inconsistencies that increase the mental burden for programmers.  I accept that that inevitably depends on individual judgement and taste.

In my judgement, the *WU system is already established and works fine for the proposed "postfix zero-extend" instructions.  If we agree that the "prefix zero-extend" instructions can't simply use *UW because of DIVUW and REMUW (if for no other reason), then something new must be invented for them.  I'm okay with either *WZX or *ZXW as a suffix, so ADDWZX or ADDZXW, etc.  Of course, it isn't entirely self-explanatory, but that goal seems out of reach to me.  It's sufficient that the "ZX" is distinctive, isn't likely to be subject to name clashes, is reasonably logical as an abbreviation of ZEXT, and is easy to remember once learned.

Clifford wrote:
> > SLLWZXI

> If at all then SLLIWZX.

Okay then.

cliffordwolf, 2019-05-24:
> If we agree that the "prefix zero-extend" instructions can't simply use *UW because of DIVUW and REMUW (if for no other reason), then something new must be invented for them.

But DIVUW and REMUW are not DIV/REM with an added UW. They are DIVU/REMU with an added W, and of course DIVU/REMU are DIV/REM with an added U.

There is no and never will be ADDU or SLLU instructions, because they would just be equivalent to ADD and SLL. So there is no possible confusion if for example ADDUW is the -W version of ADDU or the -UW version of ADD. Thus I do not agree with the statement that "something new **must** be invented for them".

Maybe you'd prefer a different scheme. And that's perfectly valid. But there surely is no imperative to use a different scheme. It's just a matter of taste. We clearly do have a choice here.

> Of course, it isn't entirely self-explanatory, but that goal seems out of reach to me.

I think "not entirely self-explanatory" is an understatement. Imo it's not a bit self explanatory. In what way is *WZX for prefix and *WU for postfix self explanatory?

I think the *UW vs *WU proposal demonstrates very well that we can at least do much better than *WZX and *WU in terms of how self-explanatory the instructions are.

jhauser-us, 2019-05-24:
> So there is no possible confusion if for example ADDUW is the -W version of ADDU or the -UW version of ADD.

We're just going to have to disagree about that.  As far as the tools and hardware are concerned, it's fine for the relationship between ADD and ADDUW to be entirely different from the relationship between DIV and DIVUW.  The computer won't care.  But it's an unwise surprise to pull on programmers.

> I think the *UW vs *WU proposal demonstrates very well that we can at least do much better than *WZX and *WU in terms of how self-explanatory the instructions are.

Since *UW doesn't cut it as self-explanatory (not if it means one thing for ADD and a different thing for DIV), I look forward to seeing your own proposal for the names.

cliffordwolf, 2019-05-25:
>> I think the *UW vs *WU proposal [..]

> Since *UW doesn't cut it as self-explanatory [..]

JFYI, while I'm trying as good as I can to stick to phrases like "I think" when describing my subjective preference you are now repeatedly presenting your subjective preference and/or personal conclusion as if it were an objective and undisputed truth. It's borderlining to begging the question. This makes it increasingly harder for me to take your position seriously, and I will undoubtedly have troubles to trust any future factual statements you make since I will have to assume that you are just presenting your feelings as if they where undisputed fact again.

As a rhetoric tactic this accomplishes the exact opposite of what you are seemingly trying to accomplish, at least when directed at me.

> I look forward to seeing your own proposal for the names.

The only possible sensible definition for an ADDU instruction that isn't equivalent to ADD would require one or both source operands to be smaller than the result. And that's _exactly_ what ADDUW is.

Therefore I see no problem with the U suffix here, and we need a size following it. Arguably there is some room for confusion regarding the final W. But that's easy enough to fix. Lets just call it `ADDU.W` then.

(If you are categorically opposed to using U here then there are 25 other letters in the alphabet that aren't U. However, I strongly feel that U makes most sense.)

jhauser-us, 2019-05-25:
I wrote:
> Since *UW doesn't cut it as self-explanatory [..]

Clifford:
> JFYI, while I'm trying as good as I can to stick to phrases like "I think" when describing my subjective preference you are now repeatedly presenting your subjective preference and/or personal conclusion as if it were an objective and undisputed truth.

Surely something can't be *self-explanatory* if it requires additional outside explanation.  If the *UW suffix means one thing for ADD and a different thing for DIV, but the difference is very logical as made clear by the documentation, then it isn't self-explanatory *by definition of the term*.  Yes, I'm going to state that as an undisputed truth.  I honestly believe that it is.

To be clear, I've never claimed we can have a totally self-explanatory naming system for the instructions.

aswaterman, 2019-05-25:
I didn't think about DIVUW when the *UW suffix was proposed.  There's an obvious and avoidable potential for confusion here.  Programmers who aren't following this discussion will understandably assume DIVUW and ADDUW are analogously defined (even though we know that doesn't make much sense, since "add unsigned then sign-extend" is exactly the same thing that the existing ADDW instruction does).

I think something like ADDZW is preferable to something like ADDU.W (with the period inserted to distinguish it from DIVUW), but I also think ADDU.W is a workable compromise.

jhauser-us, 2019-05-25:
Andrew wrote:
> I think something like ADDZW is preferable to something like ADDU.W

So you prefer ADDZW instead of ADDZXW?  Or was that just a typing error?

aswaterman, 2019-05-25:
Not sure if it was a typo or if I misread something in this thread. I think ADDZXW is the better name because it’s a bit more explicit about what the Z means.

cliffordwolf, 2019-05-26:
> I think something like ADDZW is preferable to something like ADDU.W (with the period inserted to distinguish it from DIVUW), but I also think ADDU.W is a workable compromise.

I didn't "insert a period to distinguish a it from DIVUW". I suggested ADDU.W because it makes sense.

I pointed out that ADDU describes the instruction semantic well, but that the final W is more like a size modifier, as in LR.W and CRC23C.W and FEQ.D. Should those instructions be named LRW and CRC23CW and FEQD instead as well?

If you really prefer ADDZXW over ADDU.W then you should probably also give a reason why you think this way. Otherwise there's no argument whatsoever here.

> If the *UW suffix means one thing for ADD and a different thing for DIV

Similarly ADDZW or ADDZXW could be misunderstood as *W version of some weird ADDZ or ADDZX instruction. Even if we didn't know what ADDZ or ADDZX does, we'd expect the result of the instruction to be a 32-bit value sign-extended to XLEN.

But that's not a property that this instruction has! I don't understand why you think one case is confusing and the other isn't! It just doesn't make any sense. If you reject one on the basis of the observation that it breaks an establish pattern then you have to reject the other on the same grounds.

If ADDZXW is acceptable then so must be ADDUW. If you reject ADDUW with the reasons you gave then you also must reject ADDZXW for the same reasons.

ADDU.W would be completely consistent with all established patterns afaict.

jhauser-us, 2019-05-26:
> ADDU.W would be completely consistent with all established patterns afaict.

Looking at the existing instructions (including, for example, FCVT.S.WU), seems to me it should be ADD.WU.

> Similarly ADDZW or ADDZXW could be misunderstood as *W version of some weird ADDZ or ADDZX instruction.

Of course.  But we also know it would send someone to the documentation to look up ADDZ or ADDZX, which would then protect them from misunderstanding.  Adopting a *ZXW suffix (along witih *ZXD) would block any future *ZX instructions, to avoid exactly the problem you mention.  But that seems like a small sacrifice to me.

The concern I have with ADD.WU is that it's visually so close to ADDWU, a shortcoming definitely not suffered by ADDZXW.  But if ADD.WU is really what people want, I believe it technically answers the problem raised by this issue.

cliffordwolf, 2019-05-27:
>> Similarly ADDZW or ADDZXW could be misunderstood as *W version of some weird ADDZ or ADDZX instruction.
>
> Of course. But we also know it would send someone to the documentation to look up ADDZ or ADDZX, which would then protect them from misunderstanding.

That's not a proper argument for anything. You could say the exact same thing about any other proposal. It's confusing? No problem. It will just send the user to the documentation.

> seems to me it should be ADD.WU. [..]
> The concern I have with ADD.WU is that it's visually so close to ADDWU

So you are saying it can't be ADDU.W because it has to be ADD.WU but then it can't be ADDWU?

> But if ADD.WU is really what people want

Nobody wants ADD.WU. THIS IS THE STRAWMAN YOU JUST CREATED.
AAARGH! This is so frustrating.

I'm now closing this issue. I'm so done debating informal fallacies.

If you have a proper argument I'll listen to it. But I don't have any nerves for a continuation of this BS.

closed 2019-05-27 by cliffordwolf

jhauser-us, 2019-05-27:
Clifford wrote:
> So you are saying it can't be ADDU.W because it has to be ADD.WU but then it can't be ADDWU?

I'm saying it can't be ADDWU, and ideally it shouldn't be ADDU.W.

For now, the names are as you made them, SLLIU.W, ADDU.W, and SUBU.W, and I assume they will stay that way unless they attract other comments.