riscv/riscv-bitmanip #19: brev removed from spec, still in cproofs

eroom1966, 2019-05-21:
@cliffordwolf 
Hi Clifford
I note that c.brev and c.not have been removed from the spec, but they still appear in the cproofs/insns.h
is this an oversight, or intentional as it is a pseudo-op
Thx
Lee

cliffordwolf, 2019-05-21:
> but they still appear in the cproofs/insns.h

hmmm..

I see not mention of `c.not` or even the string `not` in `cproofs/insns.h`.

The only times the string `brev` shows up in `cproofs/insns.h` is for the normal brev instruction, not c.brev.

Since `cproofs/insns.h` only model instruction semantic, not encoding, I'm not even sure what it would look like to have c.brev or c.not in that file.

eroom1966, 2019-05-21:
My mistake only mentions of brev, but it is the pseudo code anyway

GREV_PSEUDO_OP( 1, brev_p)
GREV_PSEUDO_OP( 3, brev_n)
GREV_PSEUDO_OP( 7, brev_b)
GREV_PSEUDO_OP(15, brev_h)
GREV_PSEUDO_OP(31, brev)

Thx
Lee

closed 2019-05-21 by eroom1966