riscv/riscv-bitmanip #22: Should shift immediate for SLLIU.W be 6 bits?

jhauser-us, 2019-05-27:
The document says
> `slliu.w` is identlical to `slli`, except that bits XLEN-1:32 of the `rs1` argument are cleared before the shift.

However, the proposed encoding for SLLIU.W shows another difference:  Unlike the RV64 SLLI, the shift immediate for SLLIU.W is only 5 bits, supporting a maximum shift of 31 bits.  If this was intentional, the limitation should be explained.  Otherwise, it looks like the encoding will need to be changed to accomodate a 6-bit shift for RV64 and, I presume, a 7-bit shift eventually for RV128.

cliffordwolf, 2019-05-27:
For RV64 there is no issue here: SLLIU.W with a shift amount >32 it's just SLLI.

But you raise an interesting point that I haven't considered regarding RV128. Because on RV128 SLLIU.W with a shift amount >32 is not just SLLI.

Not sure tho if that's really a problem for the intended application. Are you expecting a lot of arrays where one element is larger than 4GB? Or do you think the instruction would see a lot of use in applications other then the address calculation pattern we are targeting it at?

I think it makes sense to put SLLIU.W in OP-IMM-32 and SLLIU.D in the same spot in OP-IMM-64. We'll need to touch a new funct7 bit if we want to allow for 7-bit immediates. (There are three more left shift opcodes available with `instr[27]` set. But if we want to add the single bit ops then those encodings would already be needed for them.

cliffordwolf, 2019-05-27:
I just realized that we only need R-type *W shuffle instructions. (They perform perfect outer shuffles, so the MSB bit shouldn't change, thus leave sign extension intact. The R-type shuffle instructions are still needed for masking rs2.) So SLLIU.[WD] could use this encoding slot instead.

I'm AFK right now. I'll double check my thinking when I am back home.

jhauser-us, 2019-05-27:
> For RV64 there is no issue here: SLLIU.W with a shift amount >32 it's just SLLI.

That's a good point that I hadn't realized.  It would still be worthy of a mention in the document, assuming the encoding doesn't get changed to widen the immediate, which it sounds like it will.

> Not sure tho if that's really a problem for the intended application. Are you expecting a lot of arrays where one element is larger than 4GB? Or do you think the instruction would see a lot of use in applications other then the address calculation pattern we are targeting it at?

I agree there's a good argument to be made that we don't need even 5 bits for the immediate, since 3 or 4 bits would cover most of the expected uses.  I do expect compilers can find uses besides just address computation, such as this code:

    unsigned int m32 = ___;
    long m64 = (long)m32<<35 | ___;

But I doubt examples like this will be frequent enough to provide the justification for a full-size shift immediate.

That said, there's also a conceptual complexity to having the shift distance supported by SLLIU.W be only a subset of that of SLLI.  I'll leave it to others to decide whether that wrinkle is worth saving a few encoding bits.

cliffordwolf, 2019-05-28:
I have now removed SHFLIW and UNSHFLIW and created a new encoding for SLLIU.[WD] with space for a 7-bit immediate.

closed 2019-05-28 by cliffordwolf