riscv/riscv-bitmanip #29: rvintrin.h file name

jim-wilson, 2019-06-22:
This is a very general name.  Other extensions will also want intrinsics files.  Perhaps B extension intrinsics should be in a file with a name like rvb-intrin.h to make it clear that these are RISC-V B extension intrinsics.

If we are putting all intrinsics in one file, then we may need to conditionalize them based on whether that particular extension is enabled.  That requires a macro as per issue #28.

cliffordwolf, 2019-06-24:
Yes, the idea is put in all intrinsics in one file, and conditionalize them based on enabled extensions, similar to `x86intrin.h`.

jim-wilson, 2019-06-24:
The x86 gcc port has about 80 intrinsics files, though it does look like x86intrin.h directly or indirectly includes most of them.  So the scheme there appears to be that each major extension has its own intrinsics file, and groups of related extensions have a combined header file, and then the top level x86intrin.h includes each major group.

We can always split apart the rvintrin.h file later if it gets too big to maintain.

ARM has only 5 intrinsics files, and doesn't have a master file that includes them all.

cliffordwolf, 2019-06-24:
> The x86 gcc port has about 80 intrinsics files [..]

Note that this is just an implementation detail. The user should always include <x86intin.h>, never any of the other intrinsics header. All the other intrinsics headers have an ifdef guard like the following:

```
#ifndef _X86INTRIN_H_INCLUDED
# error "Never use <ia32intrin.h> directly; include <x86intrin.h> instead."
#endif
```

closed 2019-08-09 by cliffordwolf

MarcKarasek, 2020-06-02:
I see this file in the elf path for toolchain but not in the linux toolchain path?  
I need this for cross-compiling some bitcoin code.

jim-wilson, 2020-06-03:
I see it in my linux toolchain build.  In the install tree
rohan:2331$ find . -type f -name rvintrin.h
./lib/gcc/riscv64-unknown-linux-gnu/10.0.0/include/rvintrin.h

gcc finds it fine when I try a simple testcase.
rohan:2344$ cat tmp.c
#include <rvintrin.h>
rohan:2345$ build-install/bin/riscv64-unknown-linux-gnu-gcc -S tmp.c
rohan:2346$