riscv/riscv-bitmanip #31: gcc experimenting

jim-wilson, 2019-06-22:
I tried adding gcc optimization support for the b extension.  This is one day of work, so I only added the easy ones, didn't verify the results with execution, and haven't tried to handle every case.  The assembler is missing support for the rev and zext aliases but I can emit the pack and grevi instructions for those.  The assembler is missing support for the addwu, subwu, addu.w, subu.w, and slliu.w instructions, so those are disabled though I am able to generate them.

This patch doesn't affect dhrystone, but for coremark I see a 280 byte reduction in size, which is about 1.5%, with 99 pack instructions and 3 max instructions.  Then I realized I had the signed ee_u32 hack in my tree, so I tried undoing that.  Now I see a 384 byte reduction in size, which is about 2%, with 190 pack instructions, 3 max instructions, and 1 maxu instruction.  We can perhaps get better results with support for the missing addwu etc instructions.

[gcc-b-support.patch.txt](https://github.com/riscv/riscv-bitmanip/files/3316589/gcc-b-support.patch.txt)
[tmp.c.txt](https://github.com/riscv/riscv-bitmanip/files/3316590/tmp.c.txt)

cliffordwolf, 2019-06-24:
> The assembler is missing support for the addwu, subwu, addu.w, subu.w, and slliu.w instructions, so those are disabled though I am able to generate them.

slliu.w (and addiwu) was already implemented, but addwu, subwu, addu.w, and subu.w were not. That was an oversight/bug. I've now added them in 0dc2312.

There's spike support for those instructions, but they are one of the few yet untested instructions. (That's why I didn't catch that some of them were missing in binutils. :)

> The assembler is missing support for the rev and zext aliases but I can emit the pack and grevi instructions for those.

Yes, I haven't implemented any of the aliases yet. They, and proper support for the log2(XLEN)-1 immediate (add missing error handling for too large values), are on my todo list for binutils.

cliffordwolf, 2019-06-24:
I've now merged your gcc patch into mine, and un-commented the `(define_insn ..)` for addwu, subwu, addu.w, and subu.w, and even slliu.w (even though that doesn't seem to work yet), and added your test cases to `gcc-demo.c`, in bcb2e91.

jim-wilson, 2019-06-24:
You should be careful to keep track of authorship,  or maybe only accept patches from people known to have copyright assignments.  Otherwise, we might have trouble submitting upstream, as the FSF requires copyright assignments from all authors that contributed to a patch.  If a change is small enough and trivial enough we don't need a copyright assignment for it, but it is better to be careful up front to avoid a mess later.

cliffordwolf, 2019-06-24:
> or maybe only accept patches from people known to have copyright assignments.

That's the plan. (And I will have to get one as well.)

Btw, this is funny:

```
0000000000000048 <ctz>:
  48:	6015151b          	ctzw	a0,a0
  4c:	0005051b          	sext.w	a0,a0
  50:	00008067          	ret
```

but popcount doesn't have the extra sext.w:

```
0000000000000054 <popcount>:
  54:	6025151b          	pcntw	a0,a0
  58:	00008067          	ret
```

Also, just because I see it, I'm sure you already are aware, this andi shouldn't be needed:

```
00000000000000ac <rot>:
  ac:	01f5f593          	andi	a1,a1,31
  b0:	60b5153b          	rolw	a0,a0,a1
  b4:	0005051b          	sext.w	a0,a0
  b8:	00008067          	ret
```

jim-wilson, 2019-06-24:
I haven't tried to optimize anything yet.

The extra sext.w after ctzw is the same one we get for addw and other *w instructions.  I have an idea for how to fix it, but haven't had time to finish it.

popcount returns a value from 0 to 64, so the compiler can prove that no sign-extension is necessary when converting from 32-bit to 64-bit and optimizes away the sext.w.

clz/ctz on the other hand return an undefined (machine dependent) result if the input is zero, and this result may require sign-extension when converting it from a 32-bit to a 64-bit value.  I think some machines return -1 for instance in this case, and thus the machine independent optimizer needs to do a sign extension to convert a 32-bit -1 to a 64-bit -1.  This sign-extension should not be necessary for RISC-V, because an in-register 32-bit -1 is represented the same as an in-register 64-bti -1, but we have to feed that info back into the machine independent optimizer somehow, which we aren't doing yet.

cliffordwolf, 2019-06-25:
TL;DR: Even on x86 handling of clz/ctz is far from optimal.

> clz/ctz on the other hand return an undefined (machine dependent) result if the input is zero

jfyi, the RISC-V B extension clz/ctz _instructions_ are defined to return XLEN if the input is zero.

I find it interesting that gcc seems to treat `__builtin_clz(0)` as unspecified value. I would have expected the language "If x is 0, the result is undefined" (from gcc manual) to mean UB, not unspecified value, and always wrote code using __builting_clz accordingly. And for UB there would be no obligation for gcc to do anything reasonable in the x=0 case.

At least on x86 the following function

```
int foo(int x) { return __builtin_clz(x); }
```

is compiled to bsr (gcc 9.1.0 -O1):

```
foo(int):
        bsr     eax, edi
        xor     eax, 31
        ret
```

And the x86 ISA leaves the result of bsr undefined for a zero source operand. There isn't even a guarantee that it will always return the same value.

Also interesting, with same compiler and options:

```
int bar(int x) { return __builtin_ctz(x); }
```

```
bar(int):
        xor     eax, eax
        rep bsf eax, edi
        ret
```

Where `rep bsf` is actually `tzcnt` from BMI1, which is identical to bsf except that it returns the operand size in bits when called with a zero operand, instead of leaving this case undefined. Arguably the `rep` (which is just an encoding hack, nothing is repeated here) is not necessary given the definition of `__builtin_ctz()`.

Further, `int foobar(long x) { return x ? __builtin_ctzl(x) : 64; }` compiles to

```
foobar(long):
        xor     edx, edx
        mov     eax, 64
        rep bsf rdx, rdi
        test    rdi, rdi
        cmovne  eax, edx
        ret
```

(that's with -O3, with -Os you'll get a jump instead of the cmove), even though just `rep bsf` would be sufficient afaict. Or at least it could use the fact that BSF/TZCNT sets ZF=1 when the operand is zero. (The above code would literally just work as-is with the `test` instruction removed.) So, I guess what I'm saying is even on x86 gcc still leaves a lot on the table when it comes to clz/ctz.

Last one:

```
int boom(long x) { return __builtin_ffsl(x); }
```

This one is now using the zero flag set by BSF when the operand is zero (presumably that's just a hard-coded sequence):

```
boom(long):
        bsf     rax, rdi
        mov     rdx, -1
        cmove   rax, rdx
        add     rax, 1
        ret
```

kito-cheng, 2019-06-25:
https://gcc.gnu.org/onlinedocs/gccint/Misc.htm

Result of Clz/ctz with zero could be defined by following two marco:
```
Macro: CLZ_DEFINED_VALUE_AT_ZERO(mode, value)
Macro: CTZ_DEFINED_VALUE_AT_ZERO(mode, value)
```

cliffordwolf, 2019-06-26:
@kito-cheng I did add the following to gcc/config/riscv/riscv.h and it seems to do the right thing for `__builtin_ctzl()` but not for `__builtin_ctz()`. Am I using mode in the correct way?

```
/* Configure CLZ/CTZ behavior. */

#define CLZ_DEFINED_VALUE_AT_ZERO(MODE, VALUE) \
  ((VALUE) = GET_MODE_UNIT_BITSIZE (MODE), 2)
#define CTZ_DEFINED_VALUE_AT_ZERO(MODE, VALUE) \
  ((VALUE) = GET_MODE_UNIT_BITSIZE (MODE), 2)
```

cliffordwolf, 2019-06-26:
> gcc-b-support.patch.txt

@jim-wilson It looks like the changes for `zero_extendsidi2` and `zero_extendhi<GPR:mode>2` break zero-extend for non-B targets. I see the following ICE when building libgcc:

```
make[4]: Entering directory '/home/clifford/Work/riscv-bitmanip/patches/gcc-build/riscv64-unknown-elf/rv32i/ilp32/libgcc'
# If this is the top-level multilib, build all the other
# multilibs.
/home/clifford/Work/riscv-bitmanip/patches/gcc-build/./gcc/xgcc -B/home/clifford/Work/riscv-bitmanip/patches/gcc-build/./gcc/ -B/opt/riscv64b/riscv64-unknown-elf/bin/ -B/opt/riscv64b/riscv64-unknown-elf/lib/ -isystem /opt/riscv64b/riscv64-unknown-elf/include -isystem /opt/riscv64b/riscv64-unknown-elf/sys-include    -g -O2 -march=rv32i -mabi=ilp32 -O2  -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE  -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wstrict-prototypes -Wmissing-prototypes -Wold-style-definition  -isystem ./include   -g -DIN_LIBGCC2 -fbuilding-libgcc -fno-stack-protector -Dinhibit_libc  -I. -I. -I../../../.././gcc -I../../../../../gcc-9.1.0/libgcc -I../../../../../gcc-9.1.0/libgcc/. -I../../../../../gcc-9.1.0/libgcc/../gcc -I../../../../../gcc-9.1.0/libgcc/../include  -DHAVE_CC_TLS  -o _mulvdi3.o -MT _mulvdi3.o -MD -MP -MF _mulvdi3.dep -DL_mulvdi3 -c ../../../../../gcc-9.1.0/libgcc/libgcc2.c -fvisibility=hidden -DHIDE_EXPORTS
../../../../../gcc-9.1.0/libgcc/libgcc2.c: In function ‘__mulvdi3’:
../../../../../gcc-9.1.0/libgcc/libgcc2.c:396:1: error: unrecognizable insn:
  396 | }
      | ^
(insn 53 52 54 6 (set (reg:DI 169)
        (zero_extend:DI (reg:SI 90 [ _30 ]))) "../../../../../gcc-9.1.0/libgcc/libgcc2.c":298:6 -1
     (nil))
during RTL pass: vregs
../../../../../gcc-9.1.0/libgcc/libgcc2.c:396:1: internal compiler error: in extract_insn, at recog.c:2310
0x5518ab _fatal_insn(char const*, rtx_def const*, char const*, int, char const*)
	../../gcc-9.1.0/gcc/rtl-error.c:108
0x5518ca _fatal_insn_not_found(rtx_def const*, char const*, int, char const*)
	../../gcc-9.1.0/gcc/rtl-error.c:116
0x9a4c17 extract_insn(rtx_insn*)
	../../gcc-9.1.0/gcc/recog.c:2310
0x7b83a3 instantiate_virtual_regs_in_insn
	../../gcc-9.1.0/gcc/function.c:1605
0x7b83a3 instantiate_virtual_regs
	../../gcc-9.1.0/gcc/function.c:1975
0x7b83a3 execute
	../../gcc-9.1.0/gcc/function.c:2024
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
Makefile:498: recipe for target '_mulvdi3.o' failed
```

I've now commented out those changes in gcc.diff.

jim-wilson, 2019-06-26:
I'm travelling this weekend, but may have some time to try to fix this next week.

jim-wilson, 2019-07-07:
Part 2 of my gcc experimenting.  The 32-bit libgcc failure was due to a missing TARGET_64BIT check and once I looked I found two more.  I got slliuw working.  Primarily by adding support to use addiwu to load constants, which allows me to load the mask I need in one instruction, which then lets the gcc combine optimization pass construct the slliuw instruction from the remaining instructions.

I added sb* support, though it is a little incomplete, as the sb*iw support might be wrong.  I didn't look too closely at that; I ran out of time.  I also added support to use sbseti to load constants.

I noticed that there is no sbextiw which looks like a bug on first glance, but the fact that the result is always 0 or 1 means there is no meaningful sign-extension to do here, so probably sbextiw was left out deliberately.  I would suggest a note explaining that it is deliberately missing.  We do have both sbext and sbextw, and the only apparent difference between them is that sbext uses 6-bits for a shift and sbextw uses 5 bits for a shift.  Not clear if this is useful enough to have both instructions.

The shift-ones instructions are in the basic bit manipulation set, but this is an instruction I haven't seen before, and gcc doesn't have support for it.  Meanwhile, rotate is not in the basic bit manipulation and this is a common instruction that gcc does have support for it.  I would put rotate into the basic bit manip chapter, and pull shift-ones out, but that is a minor point.  Maybe you have some other reason for doing things this way.  Maybe rotate is harder to implement for instance.

As before, this is minimally tested and minimally optimized and could use some cleanup work.  It may not build on a 32-bit host, some expressions need to be rewritten to fix that.

I'm travelling next weekend, so the next update is probably 2 weeks away.  I will probably try to get the shift ones instructions working next time.  They can also be used to load constants.

[gcc-b-support.patch-2.txt](https://github.com/riscv/riscv-bitmanip/files/3365534/gcc-b-support.patch-2.txt)

[bitmanip-test-2.c.txt](https://github.com/riscv/riscv-bitmanip/files/3365535/bitmanip-test-2.c.txt)

cliffordwolf, 2019-07-18:
@jim-wilson Sorry for the late reply. It's been pretty busy here the last two weeks.. I've now merged your patch-2, and the fix @mablinov posted in #36.

cliffordwolf, 2019-07-18:
> so probably sbextiw was left out deliberately. I would suggest a note explaining that it is deliberately missing.

Yes, that's exactly why it's missing. I will add a note to the spec.

> the only apparent difference between them is that sbext uses 6-bits for a shift and sbextw uses 5 bits for a shift. Not clear if this is useful enough to have both instructions.

Yes, that's the only difference. I've only left out *W instructions when one can completely emulate them using the non-W instructions. I think `sbextw` is probably the most extreme case of that.

> Meanwhile, rotate is not in the basic bit manipulation and this is a common instruction that gcc does have support for it.

That's a misunderstanding. Zbb includes rol/ror/rori (second last line):

[image]

> The shift-ones instructions [..] They can also be used to load constants.

Yes, creating constant masks (usually with rs1=zero) is the main application for shift-ones.

cliffordwolf, 2019-08-09:
Closing this one to keep things tidy. Please create a new issue for the next patch.

closed 2019-08-09 by cliffordwolf

cliffordwolf, 2019-08-11:
We now have a [gcc branch](https://github.com/riscv/riscv-gcc/commits/riscv-bitmanip) to maintain our patch set. I've added the two patches from this issue in the following commits:
- https://github.com/riscv/riscv-gcc/commit/1a61a1fa993c7245d30ca018fb6e6c13b8ff7188
- https://github.com/riscv/riscv-gcc/commit/d7a6f6e016fb11c50d3ace2723e6ecd55ff02c0f

@jim-wilson Please check if the commit message and commit meta data is okay.