riscv/riscv-bitmanip #32: Possible decode error for FSRI ?

eroom1966, 2019-06-25:
@cliffordwolf 
Hi Clifford,
I wonder if you could clarify something.
In my model containing the B extensions, I am getting a decode conflict reported for FSRI during our static analysis phase, could you clarify this for me?
Here are the decode definitions

```

    DECODE_ENTRY(0, RORI,     "|01100.......|.....|101|.....|0010011|");
    DECODE_ENTRY(0, SBEXTI,   "|01001.......|.....|101|.....|0010011|");
    DECODE_ENTRY(0, SROI,     "|00100.......|.....|101|.....|0010011|");
    DECODE_ENTRY(0, FSRI,     "|.....1......|.....|101|.....|0010011|");
```

a '.' indicates a wildcard, so as you can see FSRI overlaps with RORI, SBEXTI and SROI
the FSRI[26] is part of the immediate value in the RORI, SBEXTI and SROI instructions, and 
FSRI[31:27] is rs3, but part of decode in RORI, SBEXTI and SROI

what are your thoughts ?
could this be a documentation error, and FSRI should be 
`    DECODE_ENTRY(0, FSRI,     "|.....1......|.....|101|.....|0110011|");`
not
`    DECODE_ENTRY(0, FSRI,     "|.....1......|.....|101|.....|0010011|");`

Thx
Lee

cliffordwolf, 2019-06-25:
This sentence from the spec should clarify things:

> On RV128 op[26] contains the MSB of the immediate for the shift instructions. Therefore there is no FSRI instruction on RV128. (But there is FSRIW/FSRID.)

closed 2019-06-25 by cliffordwolf

eroom1966, 2019-06-26:
Hi Clifford,
I think we are talking at cross purposes, I am referring to RV32 and RV64
These instructions overlap in the decode, for example if I had the following 2 instructions

sroi rd=r0, rs1=r0, imm=0x40 
fsri rd=r0, rs3=r4, imm=0x00

Both of these instructions would produce the same binary opcode
00100 1000000 00000 101 00000 0010011

Thx
Lee

cliffordwolf, 2019-06-26:
> I think we are talking at cross purposes, I am referring to RV32 and RV64
> These instructions overlap in the decode, for example if I had the following 2 instructions

For RV32 the shift amount is only 5 bits and RV64 the shift amount is only 6 bits.

> sroi rd=r0, rs1=r0, imm=0x40

That's not a vaild instruction in RV32 or RV64, because 0x40 is a 7 bit number.

eroom1966, 2019-06-26:
So If I understand correctly, in 
RV32 IMM>0x1f should generate an illegal operation 
RV64 IMM>0x3f should generate an illegal operation 

is that correct ?
Thx
Lee

cliffordwolf, 2019-06-27:
> So If I understand correctly, in
> RV32 IMM>0x1f should generate an illegal operation
> RV64 IMM>0x3f should generate an illegal operation
>
> is that correct ?

Yes, that's correct.

The base ISA spec is pretty explicit about this. From the RV32 instruction listing:

[image]

And the RV64 instructions listing:

[image]

Once we generate B-extension instruction listings from [riscv-opcodes](https://github.com/riscv/riscv-opcodes) we will be equally explicit about this for B-extension instructions. With the hand-written table we have right now I opted for less copy&paste.