riscv/riscv-bitmanip #34: SROW/SLOW shift masking

eroom1966, 2019-07-03:
@cliffordwolf 
Hi Clifford
In the spec the pseudo code for SROW is as follows

int shamt = rs2 & (XLEN - 1);
return ~(~rs1 >> shamt);

in the case of SROW, is XLEN the length of the machine (64 bit) or the target of the operation (32 bit)

in other words if 
rs2 = 0xFFFF_FFFF_FFFF_FFFF
is the shift amount 31 or 63 ?

eroom1966, 2019-07-03:
I think I may have answered my own question but would like it confirmed
in the case of the aboe instructions the shift mask is actually (32 - 1) which is the XLEN of the target register, not the XLEN of the machine (64 - 1)
Thx
Lee

cliffordwolf, 2019-07-03:
The spec doesn't contain any pseudo-code for SROW, but in general, the RV64 *W instructions implement the same semantics as the regular non-W version on RV32. I.e. yes, the shift mask for SROW is 31, just like the shift mask for the base ISA SLLW, SRLW, and SRAW instructions.

closed 2019-07-03 by cliffordwolf