riscv/riscv-bitmanip #36: sbclri: invalid format specifier

mablinov, 2019-07-18:
Hi all,

In Jim Wilson's proposed patch for bitmanip [found here](https://github.com/riscv/riscv-bitmanip/issues/31#issuecomment-508991101), the format specifier for `sbclri` doesn't work right because we're expecting a const_int with one bit low.

So, in order for `ctz_hwi' to work right, we need to invert the operand first, and for that we need a separate format specifier (or amend the const_int opcode in-place in the .md file, but I chose to do the former.)

The patch: 
[gcc-b-support-fix-sbclri.patch.txt](https://github.com/riscv/riscv-bitmanip/files/3406028/gcc-b-support-fix-sbclri.patch.txt)

Testcase:
```
unsigned int
f(unsigned int a)
{
        return a & ~(1 << 29);
}

```

Before:
```
f:
        sbclri  a0,a0,0
        ret

```

After:
```
f:
        sbclri  a0,a0,29
        ret

```

closed 2019-07-18 by cliffordwolf

cliffordwolf, 2019-08-11:
We now have a [gcc branch](https://github.com/riscv/riscv-gcc/commits/riscv-bitmanip) to maintain our patch set. I've added the this patch in https://github.com/riscv/riscv-gcc/commit/2d092078d52947253e29d702757f9b6c74e76134.

Please check if the commit message and commit meta data is okay.

jim-wilson, 2019-08-11:
I took a quick look at the riscv-binutils and riscv-gcc branches and they look OK to me.

With my gcc patches, I didn't do any rigorous testing, or any simulator testing, or really any testing of any kind other than verifying that I could generate the instructions with a small reasonable looking testcase.  So I'm not surprised by an error like this, and wouldn't be surprised if there are a few more like this.

commit 7e348edf (2019-07-18): Update gcc patch, fixes #36

Merged additional gcc patches from:
https://github.com/riscv/riscv-bitmanip/issues/31#issuecomment-508991101
https://github.com/riscv/riscv-bitmanip/issues/36