riscv/riscv-bitmanip #39: SLLI, SRLI, ... ROR: Confusion about encoding

mablinov, 2019-08-05:
Hi all,

I don't understand the encoding for the SLLI, SRLI, ... ROR family of instructions.

[image]

The SLLI* instruction aforementioned is not defined this way in the spec. There
is a 6-bit immediate field, not 7, as shown below for rv64i (page 30 of ISA spec v2.2):

[image]

Moreover the assembler code treats the first two fields like a funct6, followed by a 6-bit immediate:

```opcode/riscv.h:
#define OP_MASK_SHAMT		0x3f
#define OP_SH_SHAMT		20
... riscv-opc.c:
#define USE_BITS(mask,shift)    (used_bits |= ((insn_t)(mask) << (shift)))
...
      case '>': USE_BITS (OP_MASK_SHAMT,        OP_SH_SHAMT);   break;
```

Could someone please explain what's going on here? And if this is correct, should it not have it's own encoding format?

jim-wilson, 2019-08-05:
Did you try looking at the binutils patches that already exist?  You can extract the encoding info from there.  This just looks like a simple typo in the tables in the docs.

cliffordwolf, 2019-08-06:
> The SLLI* instruction aforementioned

jfyi, the `*` indicates that this is a base ISA instruction (see comment directly preceding the opcode table in the bitmanip spec document). It's not a wildcard.

> There is a 6-bit immediate field, not 7, as shown below for rv64i (page 30 of ISA spec v2.2)

The field is 7 bits on RV128, 6 bits on RV64, and 5 bits on RV32. To simplify things we just list it as a 7 bit field in our table to cover all three cases.

closed 2019-08-06 by cliffordwolf