riscv/riscv-bitmanip #40: Extending gcc-build.sh to compile g++ compiler

g-jose, 2019-08-06:
Is there a recipe available to compile g++ with support for bitmanip extension? I tried adding c++ to the --enable-languages parameter but that did not help.

jim-wilson, 2019-08-07:
You can't build a C++ compiler this way.  You need to build the C library before you can build the C++ standard library.  Building a toolchain from scratch is complicated, and requires building certain stuff in certain orders, which these build scripts are not doing.  What you can do is "make all-gcc" and "make install-gcc" in the gcc build dir, which will work, but will give you a C++ compiler with no C++ library which won't be very useful.  Technically, you also can't build a C compiler correctly this way either, but the C compiler and library interactions are simpler, and less likely to cause trouble.

I would suggest instead checking out riscv/riscv-gnu-toolchain, apply the binutils and gcc patches, and then building riscv-gnu-toolchain.   Since the binutils and gcc versions used by riscv-gnu-toolchain are not exactly the same as used by the riscv-bitmanip scripts you need to do something to work around that problem.  You can fix the patches to apply (probably hard), use git commands to check out the right branches (requires some special git knowledge), or just replace the riscv-binutils and riscv-gcc trees with the source trees created by these scripts (relatively easy).

But you might want to wait until there is some semi-official git tree or branch for this stuff.  Maintaining patch files like this is a bit of work, and not very convenient.

cliffordwolf, 2019-08-09:
What @jim-wilson said. The scripts in `patches/` are really just a means of quickly building a minimal toolchain for hacking on binutils and the gcc back-end and testing those hacks. Focus is on fast builds for quick iterations, not complete toolchain with support for multiple languages.

closed 2019-08-09 by cliffordwolf

g-jose, 2019-08-12:
Hi Jim, Clifford,
Thanks for your inputs. For now, I have changed the build settings of our software to use gcc instead and I am able to make forward progress with it. I will wait for the official commit of the support into the tool chain.
Thanks,
George