riscv/riscv-bitmanip #41: Enhanced functionality for GREV / GREVI

brucehoult, 2019-08-09:
I propose a compatible, useful, and low cost enhancement to the GREV/GREVI Generalized Reverse instructions.

At present each stage of GREV can either swap each pair of bits or else propagate them unchanged, as determined by the SHAMT bit for that stage.

I propose to perform some other function on each pair that would normally be swapped, with the same function being substituted for "swap" at each stage. The function to be performed is specified by one or more currently unused bits in rs2 or imm e.g. bit 6, or bits 6-7, or perhaps higher numbered bits to allow for 128 bit CPUs.

Supposing two bits are used, the encoding might be:

00: swap the two bits
01: both outputs are the OR of the two input bits
10: both outputs are the AND of the two input bits
11: I don't have a candidate. XOR isn't useful.

Due to DeMorgan's laws it is not necessary to provide both AND and OR, so if a useful 4th function can't be thought of then perhaps only one bit should be used.

EFFECT

I anticipate that OR or AND processing would normally be added to grev.w. grev.h, grev.b, grev.n, or grev.p. I have not evaluated whether use with other mask values is useful.

When used with one of the above masks the effect of OR instead of SWAP is to set the entire field to 1s if any bit in the field is 1. Fields consisting entirely of 0s remain as 0s.  The effect of AND is to set the entire field to 0s if any bit in the field is 0. Fields consisting entirely of 1s remain as 1s.

For example, with an input of cbf20097147200ac the output of GREV.B.OR is ffff00ffffff00ff.

APPLICATIONS

If the output of the above GREV.B.OR is inverted to 0000ff000000ff00 then CLZ or CTZ can be used to determine the position of the first or last zero byte in the input value.

Alternatively, the input value could be inverted and then GREV.B.AND will produce the necessary input for CLZ or CTZ.

This is very valuable in efficient implementation of C string processing functions such as strlen(), strcpy(), strcmp().

Along with general benefits, this will provide a large boost to RISC-V scores in Dhrystone.

Using GREV.H.OR provides the same functionality for UTF-16 or UCS-2, or GREV.W.OR for UTF-32 or UCS-4.

COST

GREV is dominated by wire cost. The logic at each node is extremely small and increasing its size will not meaningfully impact the cost of GREV in either SoC or FPGA.

In particular, in FPGAs with splitable 6-LUTs we have five inputs (the two input bits, the SHAMT swap/fn enable for the stage, and my proposed two function select bits) and these inputs determine two independent bit outputs -- a perfect fit.

The attached program provides a reference C implementation of the proposed modification, checks that it produces the same output as the existing reference implementation when the function select bits are 0, and demonstrates finding all-zero fields of widths 4 to 32.

[grev.txt](https://github.com/riscv/riscv-bitmanip/files/3484912/grev.txt)

cliffordwolf, 2019-08-09:
Interesting. Below are my thoughts. I'll also send an email to the task group mailing list asking for feedback from others in the group.

First, even though this would best be implemented as part of the GREV circuit, I think it should be a different instruction, not controlled by additional bits in the shift amount. I also think only the OR version would be more than sufficient. (As you have mentioned, the AND version would be implemented easily with inverting before and after.) I'll call that instruction GROC (Generalized OR-Combine) here. Suggestions for better names are of course welcome.

GREV.B.OR would indeed be useful for strlen(), strcpy(), strcmp(). But it is equivalent to `bmator(x, -1L)`. Furthermore, a single-stage GROC can easily be emulated by running single-stage GREV followed by OR. Thus, GREV.H.OR would simply be `bmator(x, -1L)`, followed by `rev8.h+or`.

Do you know any other applications that looking for zero bytes (or half-words or words) for speeding up strlen(), strcpy(), strcmp() and alike? Because if that's the only use-case then a dedicated instruction for counting trailing nonzero bytes might be cheaper, both in terms of encoding space and hardware cost.

brucehoult, 2019-08-09:
bmator is an RV64-only instruction. Most of the people who care about Dhrystone, for example, are using RV32.

I was originally thinking about this as a stand-alone instruction, but worried about the opcode space issues. If GREV was for some reason rejected then that would be an option. As it stands, I feel it provides more evidence to include GREV :-)

brucehoult, 2019-08-09:
Note that it's not only for looking for zero bytes. If you want to look for something else just XOR with it first, and then look for zero.

cliffordwolf, 2019-08-09:
> bmator is an RV64-only instruction. Most of the people who care about Dhrystone, for example, are using RV32.

That's a good point!

Also: Zbm is much more expensive than Zbp in terms of hardware cost, therefore I would expect it to be less common, even in RV64 implementations. And implementors will likely opt for multi-cycle implementations of `bmat[x]or` but single-cycle implementation of Zbp.

> Note that it's not only for looking for zero bytes. If you want to look for something else just XOR with it first, and then look for zero.

Well, yes, I am aware. But you can do the same XOR thing with an instruction that counts trailing nonzero bytes.

cliffordwolf, 2019-08-10:
I've done some preliminary experiments and I agree that the HW overhead for this instruction would be very small, assuming existing GREV and [UN]SHFL support.

Btw, another usecase would be duplicating parts of a word, when the rest of the word is zero initialized. For example `gorc(0xab, 24) => 0xabababab`. Or, using the same naming scheme for instruction aliases as for `grev`, `orc8(0xab) => 0xabababab`.

cliffordwolf, 2019-09-04:
GORC has been added in bitmanip v0.91. Closing this issue now.

closed 2019-09-04 by cliffordwolf