riscv/riscv-bitmanip #43: Clarification on cmix documentation

deepsrc, 2019-08-19:
I need a small clarification. The description for cmix instruction says that -

> It is equivalent to the following sequence.
> and rd, rs1, rs2
> andn t0, rs3, rs2
> or rd, rd, t0

Is it implied that the register t0 will be modified as a result of the execution of this instruction?

cliffordwolf, 2019-08-19:
> Is it implied that the register t0 will be modified as a result of the execution of this instruction?

No. I've now replaced the sentence "It is equivalent to the following sequence" with "It replaces sequences like the following". I hope that makes it clearer. (Will be in the PDF next time I update it.)

closed 2019-08-19 by cliffordwolf

deepsrc, 2019-08-19:
Thanks for the clarification