riscv/riscv-bitmanip #48: Consider changing BFP not to wrap around new field in result

jhauser-us, 2019-10-07:
As instruction BFP is currently defined, the bit field it overlays in the rs1 value may wrap around to span both the high (most-significant) and low (least-significant) ends of the result,  For example, this sequence,

    li t0,12<<24|26<<16|0xABCD
    bfp t1,zero,t0

for RV32, leaves `t1` with the value `0x340002AF`, because the 16-bit value `0xABCD` shifted left 26 bits (without clipping) is `0x2AF34000000`, and this value wraps around from high bits to low bits in the result.

Are there expected advantages to this wrapping?  My analysis indicates that the hardware for BFP (and for the B extension generally) can be a little reduced by not defining BFP to wrap around this way.  The basic reason is because BFP requires the hardware to separately create a mask in addition to shifting (or rotating) rs2[15:0], and forcing this mask to wrap around adds a little extra circuitry.

I can imagine some applications might benefit from wrapping around, while others benefit more from *not* wrapping around.  If there are good reasons to prefer wrap-around, I suggest adding an explanation of that choice to the document.  If not, I propose modifying the specified behavior to use shifts instead of rotations, like so:

    uint_xlen_t bfp(uint_xlen_t rs1, uint_xlen_t rs2)
    {
        int len = (rs2 >> 24) & 15;
        int off = (rs2 >> 16) & (XLEN-1);
        len = len ? len : 16;
        uint_xlen_t mask = slo(0, len) << off;
        uint_xlen_t data = rs2 << off;
        return (data & mask) | (rs1 & ~mask);
    }

For my example above, the value left in `t1` would then be `0x34000000`.

(Note that, the hardware could still quietly substitute

    uint_xlen_t data = rol(rs2, off);

for computing `data` without changing the behavior, if that's more convenient.  The issue is just with the rotation of the mask.)

cliffordwolf, 2019-10-14:
> The basic reason is because BFP requires the hardware to separately create a mask in addition to shifting (or rotating) rs2[15:0], and forcing this mask to wrap around adds a little extra circuitry.

That's at least not how I built it. See https://github.com/riscv/riscv-bitmanip/blob/master/verilog/rvb_shifter/rvb_shifter.v.

In a previous discussion on the bitmanip mailing list it was concluded that in most cases (even without funnel shifts), the most efficient way of implementing the shifter would be a `2*XLEN` rotate shift. So my BFP implementation is based on that: Set the `2*XLEN` shift value to `{111...111, din_rs2[15:0] | bfp_mask, 000...000, din_rs2[15:0] & ~bfp_mask}` where `bfp_mask[15:0] = 16'h FFFF << bfp_len`, then rotate shift by `bfp_off`, then set `Y = ((X | Z) & A) | (X & Z)` (with `A = rs1` and `X` and `Z` being the two halves of the 2*XLEN result).

I'd have to run a more detailed analysis to assess the difference in required HW for the two definitions.

> Are there expected advantages to this wrapping?

Placing a bit field >=(XLEN-16) bit in the middle of a word is the same as shifting that bit field value up and then placing the "surrounding" using a wrapping BFP.

jhauser-us, 2019-10-14:
> I'd have to run a more detailed analysis to assess the difference in required HW for the two definitions.

Thanks for pointing me to your specific Verilog implementation (which no doubt was there all along, but I forgot to look).  My analysis so far is built on some different assumptions, such as no support for the ternary instructions, for lack of a third register read port.  I hope to provide more feedback eventually about which BFP version is likely to be smaller in hardware.  But first, back to the motivation question:

> > Are there expected advantages to this wrapping?
>
> Placing a bit field >=(XLEN-16) bit in the middle of a word is the same as shifting that bit field value up and then placing the "surrounding" using a wrapping BFP.

Could you show an example where this way of using BFP is actually beneficial?  When I try it, the effort to construct the rs2 operand for the BFP is too much work.

Assuming RV32, consider the problem of injecting a 17-bit value from `a1` into bits [21:5] of `a0`, leaving the result in `t0`.  When I try to use BFP as you suggest, I get a sequence like this:

    rori t0,a0,5
    srli t0,t0,17
    li t1,15<<24|22<<16
    or t0,t1,t0
    slli t1,a1,5
    bfp t0,t1,t0

A more obvious sequence, using only existing RV32I instructions, is actually shorter by one instruction:

    li t0,~(0x1FFFF<<5)    ; expands to 2 instructions
    and t0,a0,t0
    slli t1,a1,5
    ori t0,t0,t1

Is there a better way to use BFP than what occurred to me?

jhauser-us, 2019-10-14:
> Assuming RV32, consider the problem of injecting a 17-bit value from `a1` into bits [21:5] of `a0`, leaving the result in `t0`.

Using rotates and shifts, I found I could do it in only 4 instructions:

    roli t0,a0,10
    slli t0,t0,17
    or t0,t0,a1
    roli t0,t0,5

Or, if we need to clip the value from `a1`, 5 instructions:

    rori t0,a0,5
    srli t0,t0,17
    slli t1,a1,15
    or t0,t0,t1
    rori t0,t0,10

Note that versions of these sequences (in the same number of instructions) will work for both RV32 and RV64, for all field widths, not just widths >= XLEN - 16.  This definitely beats my attempt to use the wraparound BFP.

jhauser-us, 2019-10-14:
I wrote:
> My analysis indicates that the hardware for BFP (and for the B extension generally) can be a little reduced by not defining BFP to wrap around this way. The basic reason is because BFP requires the [...]

Clifford Wolf:
> That's at least not how I built it.

Because your hardware implements the funnel shift ternary instructions, you have a 128-bit rotating shifter.  For BFP, you're taking advantage of the double-wide rotation to get the effect of shifting the mask "for free" (although your technique is a little more subtle than that).  However, for any implementation that doesn't have the ternary instructions, the main shifter need be only 64 bits.  In that case, supporting BFP requires also being able to generate the mask separately, as I mentioned.

I've attached a modified version of your `rvb_shifter.v` that I believe eliminates the wrap-around for BFP and should also be a little smaller in hardware than your current one.  Unfortunately, I haven't had time to test it, because I don't yet have the same setup as you.  But I have reasonably high confidence it will work.  If it doesn't, let me know, and I'll invest the energy to fix it.

[rvb_shifter.zip](https://github.com/riscv/riscv-bitmanip/files/3726068/rvb_shifter.zip)

cliffordwolf, 2019-10-15:
> Or, if we need to clip the value from a1, 5 instructions:
> 
> ```
> rori t0,a0,5
> srli t0,t0,17
> slli t1,a1,15
> or t0,t0,t1
> rori t0,t0,10
> ```

Also 5 instructions with BFP (result in a0, so it looks like the body of a C function):

```
slli a1, a1, 5
rori a2, a0, 22
addi a0, zero, -234
pack a0, a2, a0
bfp a0, a1, a0
```

But the `addi+pack+bfp` sequence is a fuse-able sequence. So with macro-op fusion it's *three macro-ops*. (See "2.13.3 Fused *-bfp sequences" in current draft spec.)

On RV64 we would use `packw` instead of `pack`. Otherwise it's the same sequence.

cliffordwolf, 2019-10-15:
> My analysis so far is built on some different assumptions, such as no support for the ternary instructions, for lack of a third register read port.

I was not assuming funnel shifts or ternary instructions either. As I wrote above:

> In a previous discussion on the bitmanip mailing list it was concluded that in most cases **(even without funnel shifts)**, the most efficient way of implementing the shifter would be a 2*XLEN rotate shift.

When implementing only Zbb then it possibly would make sense to implement the shifter differently, because other implementations provide better opportunities for sharing resources with rev8, rev, and orc.b. (I have writing a small 32-bit Zbb unit that demonstrates this on my todo list.)

But I'd argue that just "Zbb" + "Zbf" is an odd combination. I think it's more likely that a core implementing "Zbb" would do that as part of "B".

> I've attached a modified version of your rvb_shifter.v that I believe eliminates the wrap-around for BFP and should also be a little smaller in hardware than your current one.

Thanks. I'll have a look.

cliffordwolf, 2019-10-15:
> Unfortunately, I haven't had time to test it, because I don't yet have the same setup as you. But I have reasonably high confidence it will work. If it doesn't, let me know, and I'll invest the energy to fix it.

It doesn't seem to implement BFPW on RV64 in accordance to your definition of `bfp`.

```
testdata_4.hex: idx=001 insn=0x0800703b rs1=0x7d312fe94148526f rs2=0x2bb76a4d192b8f72 rs3=0x707a991048499f3c rd=0x00000000414ffa6f expected=0x00000000414b926f ERROR
Makefile:64: recipe for target 'testbench_4.ok' failed
```

Same for BFP on RV32:

```
testdata_5.hex: idx=009 insn=0x08007033 rs1=0x00000000fead627b rs2=0x000000001a214e1d rs3=0x0000000010cc4f94 rd=0x00000000fead67ff expected=0x00000000fead643b ERROR
Makefile:68: recipe for target 'testbench_5.ok' failed
```

Only BFP on RV64 seems to be correct.

jhauser-us, 2019-10-15:
> It doesn't seem to implement BFPW on RV64 in accordance to your definition of `bfp`.

So sorry.  I logged on early this morning because I realized I forgot one little detail for the 32-bit case.  Try this one:
[rvb_shifter.zip](https://github.com/riscv/riscv-bitmanip/files/3729344/rvb_shifter.zip)
Fingers crossed.

cliffordwolf, 2019-10-15:
> Suffice it to say, I disagree.

Do you have data / reference designs that support your position? Back when we had this discussion on the mailing list we benchmarked various designs for shifters and found that this one is the most efficient. (Don't use Yosys to benchmark this. We've found that Yosys misses some optimizations that other tools don't, causing the results to swing it the wrong direction sometimes.)

> then perhaps that will be sufficient argument for making the change

The main argument for the rotating BFP behavior remains the same: 3 macro-ops for placing a field >16 bits. (see above)

> Try this one: rvb_shifter.zip Fingers crossed.

BFPW on RV64:

```
testdata_4.hex: idx=044 insn=0x0800703b rs1=0x3a0197f21705a22d rs2=0xcb4a222d6ef770ea rs3=0x9da4637f28baf8da rd=0x000000007505a22d expected=0x000000007505a238 ERROR
Makefile:64: recipe for target 'testbench_4.ok' failed
```

BFP on RV32:

```
testdata_5.hex: idx=026 insn=0x08007033 rs1=0x00000000b74c3d2a rs2=0x0000000016ded635 rs3=0x00000000b2df0d45 rd=0x00000000774c3d2a expected=0x00000000774c3d2d ERROR
Makefile:68: recipe for target 'testbench_5.ok' failed
```

jhauser-us, 2019-10-15:
For those last two reported errors, it looks to me like my unit is delivering the correct result without wraparound, while the "expected" result incorrectly includes the wraparound.

> Do you have data / reference designs that support your position?

I do, but not that I can share.

Let's just assume I'm wrong about that, if there's nobody to back me up.  By modifying your `rvb_shifter`, I'm also attempting to show that, even under your own assumptions, the non-wraparound BFP can be a little more efficient.

> The main argument for the rotating BFP behavior remains the same: 3 macro-ops for placing a field >16 bits.

That's assuming a 3-instruction fusion of ADDI + PACK + BFP.  Forgive me, but I have to think machines that support that fusion will be rare at best.

As long as we're assuming aggressive fusion, why can't the competition fuse instructions too?  Let's say I solve my original example problem with this code:

    lui t0,0xFFC00   
    addi t0,t0,0x1F   
    and t0,a0,t0     
    slli a0,a1,15    
    srli a0,a0,10
    or a0,t0,a0

Looks to me that the first three instructions and the last three are each fuse-able sequences, giving me only *2 macro-ops*, no wraparound BFP needed.

For this hypothetical RV32 machine, not only would the 6-instruction sequence above be documented as the best idiom for bit-field insertion, we could also expect the two fuse-able sequences (and some others like them) to be targeted by the compiler for other purposes, making them more generally useful than your BFP sequence, I would think.

For RV64, I can conjure a similar scheme, sometimes 2 macro-ops and sometimes 3.

Can you prove this isn't a better solution?  In my opinion, your justification for the wraparound BFP hangs on a very thin thread.

cliffordwolf, 2019-10-16:
> I do, but not that I can share.

That's unfortunate. Can you share what kind of shifter architecture it is? Maybe it is a kind that we did not evaluate.

> That's assuming a 3-instruction fusion of ADDI + PACK + BFP. Forgive me, but I have to think machines that support that fusion will be rare at best.

The whole idea of BFP is that, instead of standardizing an 48-bit or 64-bit instruction, we standardize an instruction that can be used in either a fuse-able ADDI+PACK+BFP sequence or a fuse-able LUI+ADDI+BFP sequence, because the feedback I got from the community is that they rather fuse three instructions than decode one long instruction.

So I'm not assuming a machine here that aggressively fuses any fuse-able three instruction sequence. Just one that fuses the two sequences BFP was specifically built for.

I'm looking at the above error message now and post again when I have a reply there.

> Can you prove this isn't a better solution? In my opinion, your justification for the wraparound BFP hangs on a very thin thread.

I concede that it doesn't hang on a steel wire.  :D

From a point of just instruction semantics I prefer the warp-around version because the non-wrapping version can always be emulated by setting a shorter length. So I think the resource difference must either be significant, or we must be very certain that giving up on the wrapping behavior doesn't mean giving up on a perfectly valid use-case of the wrapping version.

cliffordwolf, 2019-10-16:
I must have somehow f***ed up the insn.h change for testing against your bfp semantics when I ran it yesterday. I re-ran the test today and now it's looking fine. Weird. Transistor counts:

| module                        | unmodified | modified | rel change |
|-------------------------------|--------:|--------:|---------:|
| rvb_shifter_xlen32_sbop0_bfp0 |   3036 |   3044 |  +0.26% |
| rvb_shifter_xlen32_sbop0_bfp1 |   5840 |   5886 |  +0.78% |
| rvb_shifter_xlen32_sbop1_bfp0 |   4098 |   4036 |  -1.51% |
| rvb_shifter_xlen32_sbop1_bfp1 |   7070 |   7048 |  -0.31% |
| rvb_shifter_xlen64_sbop0_bfp0 |   7484 |   7434 |  -0.66% |
| rvb_shifter_xlen64_sbop0_bfp1 |  12934 |  12630 |  -2.35% |
| rvb_shifter_xlen64_sbop1_bfp0 |   9458 |   9446 |  -0.12% |
| rvb_shifter_xlen64_sbop1_bfp1 |  15810 |  15166 |  -4.07% |

As you can see the noise floor in this test is quite high. (-1.51% area change in a version that doesn't even support BFP.) But it looks like your version really is a tiny bit smaller, but only on RV64. On RV32 it's on average even a bit larger (but I would argue it's just noise either way in the RV32 case).

In your shifter architecture, with your tools, is the difference there also in that 2%-4% neighborhood, or is it larger?

jhauser-us, 2019-10-16:
> From a point of pure instruction semantics I prefer the warp-around version because the non-wrapping version can always be emulated by setting a shorter length.

Yes, I had that same thought as well.  But now you also have a taste of what I'm seeing on the cost side of the ledger.  Thanks for running the experiment.

Given my different design, without funnel shifts, the baseline size of my shifter unit is smaller.  The relative delta I get for implementing the wraparound BFP is correspondingly higher,  more consistently in the 5% range, give or take.

I suspect the reason you're not seeing the same effect for XLEN = 32 is because you're having the tools optimize down the 64-bit `rvb_shifter`, and there's some overhead remaining from that process.  My guess is, if you created pure 32-bit versions of the same units, yours and mine, the difference would show up clearly again.

> The whole idea of BFP is that, instead of standardizing an 48-bit or 64-bit instruction, we standardize an instruction that can be used in either a fuse-able ADDI+PACK+BFP sequence or a fuse-able LUI+ADDI+BFP sequence  [...]

This seems logical, but I'm afraid it may be an unintentional false promise from the developers.  Because, although they may start off with plans to implement the fusions you say, at some point during the long development process it's almost certain to occur to them that fusing other three-instruction sequences will give a bigger payoff.  So one thing I'm pretty sure of, the only machines to fuse those BFP sequences will be ones that already fuse many other three-instruction sequences.  Which leaves me expecting only a small fraction of high-end RISC-V machines (if any).

That's my take.

cliffordwolf, 2019-10-18:
I've made some changes to the `rvb_shifter` core and now I'm at -5% for the largest version, with unchanged BFP semantics.

This is not to say that the unchanged semantics necessarily has lower area, but it shows that there is a not of noise involved in these measurements.

| module                        |    old |    new | rel change |
|-------------------------------|-------:|-------:|--------:|
| rvb_shifter_xlen32_sbop0_bfp0 |   3036 |   3030 |  -0.19% |
| rvb_shifter_xlen32_sbop0_bfp1 |   5840 |   5848 |  +0.13% |
| rvb_shifter_xlen32_sbop1_bfp0 |   4098 |   4088 |  -0.24% |
| rvb_shifter_xlen32_sbop1_bfp1 |   7070 |   7004 |  -0.93% |
| rvb_shifter_xlen64_sbop0_bfp0 |   7484 |   7338 |  -1.95% |
| rvb_shifter_xlen64_sbop0_bfp1 |  12934 |  12974 |  +0.30% |
| rvb_shifter_xlen64_sbop1_bfp0 |   9458 |   9456 |  -0.02% |
| rvb_shifter_xlen64_sbop1_bfp1 |  15810 |  14998 |  -5.13% |

> I suspect the reason you're not seeing the same effect for XLEN = 32 is because you're having the tools optimize down the 64-bit rvb_shifter, and there's some overhead remaining from that process.

I don't think so. The new version of my core now has an explicit 32-bit version of the datapath and the core size for 32 bit is virtually unchanged.

I think it's because the only real optimization in your version is that it gets rid of `wire [63:0] XZ = {Z[63:32], wmode ? X[63:32] : Z[31:0]};`, and that operation was free anyway for a 32 bit core because there `wmode` is constant 1.

> This seems logical, but I'm afraid it may be an unintentional false promise from the developers. [..] So one thing I'm pretty sure of, the only machines to fuse those BFP sequences will be ones that already fuse many other three-instruction sequences. Which leaves me expecting only a small fraction of high-end RISC-V machines (if any).

Sure, no machine will add instruction fusion just for BFP. But that wasn't the argument anyway.

But I strongly disagree with the "if any" portion of your comment. RISC-V cores that fuse macro ops already exist.

I also disagree with the quantifier in "already fuse many other three-instruction sequences". I'd expect a core that can fuse _any_ three-instruction sequence and supports BFP to also fuse the BFP sequences.

And pretty much everyone I talked to who said they support macro-op fusion or are planning on supporting macro-op fusion told me that they have macro-ops of up to three instructions in mind.

jhauser-us, 2019-10-18:
> I've made some changes to the `rvb_shifter` core and now I'm at -5% for the largest version, with unchanged BFP semantics.

Okay, now here's my modification of your update, again changing BFP to not wrap around:
[rvb_shifter-3.zip](https://github.com/riscv/riscv-bitmanip/files/3745393/rvb_shifter-3.zip)
Actually, I'm providing two versions for you to try this time.  (And as before, I'm still doing this blind, without testing, so I apologize in advance if they don't pass tests right away.)

> I'd expect a core that can fuse any three-instruction sequence and supports BFP to also fuse the BFP sequences.

You mean, if a core supports BFP and can fuse *any* three-instruction sequence, it will fuse the three-instruction BFP sequences?  I happen to know of a core in development for which that won't be true.  (But perhaps it's an outlier; I don't know.)

> Pretty much everyone I talked to who said they support macro-op fusion in their core or are planning to support macro-op fusion told me that they have macro-ops of up to three instructions in mind.

In which case maybe (or maybe not) the 2-macro-op example I gave earlier is supported, trumping the 3-macro-op BFP sequence.  The salient point being, it's hard to predict this right now.

The only example use case we have so far for wraparound BFP provides a small benefit for only a minority of bitfield insertions (width >= XLEN - 16, definitely a minority for RV64, and likely so for RV32 too) and also depends on some assumptions that may or may not pan out.  I claim I'm seeing a cost of about 5% in the size (and maybe delay) of the shifter, at least for some implementations.  For comparison, can the expected benefit even be measured with confidence, given the future unknowns involved?

jhauser-us, 2019-10-18:
> (And as before, I'm still doing this blind, without testing, so I apologize in advance if they don't pass tests right away.)

I've gotten now so I can run `make test` within `verilog/rvb_shifter`.  Both of my new modified `rvb_shfiter` versions seemed to pass (with appropriate `cproofs/insns.h`, of course), so hopefully that's a good sign they will for you too.

cliffordwolf, 2019-10-19:
> Actually, I'm providing two versions for you to try this time.

Ok, I love the results for your 3b version.

| Variant                       | orig  |  3a   |  3b   | rel 3a| rel 3b |
|-------------------------------|------:|------:|------:|------:|-------:|
| rvb_shifter_xlen32_sbop0_bfp0 |  3030 |  3038 |  3016 | +0.26 |  -0.46 |
| rvb_shifter_xlen32_sbop0_bfp1 |  5848 |  5718 |  4846 | -2.22 | -17.13 |
| rvb_shifter_xlen32_sbop1_bfp0 |  4088 |  4218 |  4094 | +3.18 |  +0.15 |
| rvb_shifter_xlen32_sbop1_bfp1 |  7004 |  6620 |  6108 | -5.48 | -12.79 |
| rvb_shifter_xlen64_sbop0_bfp0 |  7338 |  7386 |  7436 | +0.65 |  +1.34 |
| rvb_shifter_xlen64_sbop0_bfp1 | 12974 | 12898 | 10214 | -0.59 | -21.27 |
| rvb_shifter_xlen64_sbop1_bfp0 |  9456 |  9530 |  9470 | +0.78 |  +0.15 |
| rvb_shifter_xlen64_sbop1_bfp1 | 14998 | 14970 | 12270 | -0.19 | -18.19 |

 It's counter-intuitive to me why `<< bfp_off` is so much more efficient than widening the main shifter by adding that `Z` signal. My expectation would have been that for 32-bit the Z signals adds 16+24+28+30+31 = 129 MUXes and that `{din_rs2[15:0] & ~bfp_mask} << bfp_off` would add 17+19+23+31+32 = 122 MUXes, almost the same. It looks like the other optimizations are making the big difference here (sharing of `A&~X` with `sbclr`, simplified `bb`, etc).

I will send an email to the bitmanip mailing list asking if anyone has a good objection to removing the rotate behavior from BFP.

allenjbaum, 2019-10-21:
Question (not having looked at all the detailed verilog): do the cost estimates for BFP include also handling all the other types of shifts and rotates (so, there is a single shifter/rotator that handles all the various ops)? 

The example given for wrapping BFP is a bit contrived, in my opinion; I  am suspicious when one of the operands (in this case, the the immediate in bfp_mask) is a fixed value for all the useful cases.

jhauser-us, 2019-10-21:
> do the cost estimates for BFP include also handling all the other types of shifts and rotates (so, there is a single shifter/rotator that handles all the various ops)?

Yes, all supported shift operations are covered by a single shifter unit.  Although in my case, only the two-input functions are supported, so the funnel shifts aren't included.  Clifford considers four different cases for both RV32 and RV64: with/without the single-bit functions (SBSET, etc.), and with/without BFP.  He always includes the funnel shifts, together with all other shift operations (SLL, SRL, SRA, etc.).

allenjbaum, 2019-10-21:
Once the rotate ops are implemented, all the other ops come almost for free
(except for reading the 3rd operand of the funnel shift).
I'm not convinced that generating a mask that wraps or doesn't  is even
significant; it just changes a high/lo mask AND into a high/lo mask XOR.
Not that I'm convinced that wraparound is necessary :-)

jhauser-us, 2019-10-21:
> I'm not convinced that generating a mask that wraps or doesn't is even significant; it just changes a high/lo mask AND into a high/lo mask XOR.

But it's not free, and, begging your pardon, I've given an estimate of what the cost is for me.  Show me there's a compelling reason to implement the wraparound, and I'll happily drop my objection to it.

allenjbaum, 2019-10-21:
I think we probably have a different threshold of "cost". It may be 50% or
a core, but a vanishingly small percentage of an entire SOC.
In any case, I'm perfectly happy to not have wraparound for this.
Its just that I don't think the cost should be the main focus; I think use
cases should be. Just my opinion, and note that I don't have biases
otherwise.

jhauser-us, 2019-10-21:
In terms of area costs alone, I can agree.  But an accumulation of excess transistors in the wrong places can impact the maximum clock speed, which has a more global impact.  And the integer ALUs are just such a place.  The transistors you have to have, you pay the cost, but it's wise to exclude whatever you can.  It's conceivable (though not guaranteed) that a 5% savings in every critical unit could add up to a 5% faster clock speed for the core.

allenjbaum, 2019-10-21:
In my experience, ALUs are almost never  the critical path.
Branch resolution, TLB translation, and especially Cache/SRAM access,  -
lots of other functionality, but ALUs - rarely.

cliffordwolf, 2019-10-23:
> In any case, I'm perfectly happy to not have wraparound for this.
Its just that I don't think the cost should be the main focus; I think use
cases should be. Just my opinion, and note that I don't have biases
otherwise.

I have a hard time producing convincing usage examples for the wraparound behavior. So I'm okay with dropping it. (But I still think it is the "nicer" instruction semantic. :)

However, you raise an interesting point. The reason why even on RV64 we only allow 16-bit to be placed is to safe chip area, but supporting 32-bit does not actually add that much in terms of area. So maybe that is overly conservative and a premature optimization.

So, at least for now, I think we should go with the following definition for BFP (and new BFPW):

```
uint_xlen_t bfp(uint_xlen_t rs1, uint_xlen_t rs2)
{
        uint_xlen_t cfg = rs2 >> 16;
        if (XLEN > 32) {
                cfg = rs2 >> (XLEN/2);
                if ((cfg >> 30) == 2) 
                        cfg = cfg >> 16;
        }
        int len = (cfg >> 8) & (XLEN/2-1);
        int off = cfg & (XLEN-1);
        len = len ? len : XLEN/2;
        uint_xlen_t mask = slo(0, len) << off;
        uint_xlen_t data = rs2 << off;
        return (data & mask) | (rs1 & ~mask);
}
```

The behavior on RV32 is unchanged. On RV64 this will take the config from `rs2[63:48]` when `rs2[63:62] == 2` and from `rs2[47:32]` otherwise. This allows the use of LUI to construct the config for the static case, and PACKH when it comes from two registers.

Synthesis results:

```
rvb_shifter_xlen32_sbop0_bfp0          3024
rvb_shifter_xlen32_sbop0_bfp1          4816
rvb_shifter_xlen32_sbop1_bfp0          4048
rvb_shifter_xlen32_sbop1_bfp1          6034
rvb_shifter_xlen64_sbop0_bfp0          7362
rvb_shifter_xlen64_sbop0_bfp1         11804
rvb_shifter_xlen64_sbop1_bfp0          9502
rvb_shifter_xlen64_sbop1_bfp1         13678
```

This is the patch: 
[bfp64_diff.txt](https://github.com/riscv/riscv-bitmanip/files/3760838/bfp64_diff.txt)

cliffordwolf, 2019-10-23:
> The behavior on RV32 is unchanged.

By that I mean unchanged with respect to what @jhauser-us  is suggesting. This is still dropping the wrap-around behavior.

cliffordwolf, 2019-10-23:
I have now put this in a branch of riscv-bitmanip: https://github.com/riscv/riscv-bitmanip/tree/bfp64

closed 2020-03-13 by cliffordwolf