riscv/riscv-bitmanip #51: Benchmark - Variable Length Unary Integer Coding (clz/ctz)

michaeljclark, 2019-11-08:
This might be useful:

- https://github.com/michaeljclark/vlu

VLU(...) is a little-endian variable-length integer coding that prefixes data bits with unary code length bits. The length is recovered by counting the least significant set bits, which encode a count of n-bit basic units. The data bits compactly trail the unary code prefix.

- encode uses `clz`
- decode uses `ctz`

With an 8 bit basic unit, the encoded size is similar to LEB128; 7-bits can be stored in 1 byte, 56-bits in 8 bytes and 112-bits in 16 bytes. Decoding, however, is significantly faster than LEB128, as it is not necessary to check for continuation bits every byte, instead the length can be decoded in a single count bits operation.

While VLU is not in major use, it could be substituted where LEB128 is used with reasonably significant benefits depending on the frequency of variable length fields. LEB128 probably performs similarly to VLU on a machine without bit scan forward and reverse. There are also potential SIMD or vector optimisations. For example, a decoder could have a predictor, and switch from a "per field" mode to a set of optimized modes. e.g. 128-bit SIMD code for parallel decoding of 16 x 7-bit fields.

There is symmetry in the encode and decode, with `clz` for figuring out the size of a word, and `ctz` to read the /prefix/ from the little-end. The code is a pretty good example of why little-endian makes more sense. The benchmarks currently perform decoding of 8-bit through to 56-bit and there is an optimized decoder for x86-64 BMI. I am investigating x86 SIMD and want to add support for big numbers. 112-bits and >= 128-bits.

michaeljclark, 2019-11-08:
FYI - The array encoder and decoder uses misaligned loads and stores.

I have a branch, that I have not pushed yet, for array encoding and decoding, that does a periodical refill and swap of two 64-bit words, combined with SHIFT-OR-SHIFT to reassemble extracted variable length fields. It's about 1.25-1.95 times faster, on x86, to use the misaligned load and L1 cache, to extract words of variable length at byte offsets. It proves to me that a byte code is slightly faster than a bit code, for this VLU code at least.

I'll push the branch. It is possible there are some optimizations that I have missed. Also, the results could be different on different micro-architectures. Aligned load with bit shifts will be better on some RISC-V implementations that fix up misaligned loads and stores in firmware.

I can push the branch if there is interest. It seems to indicate that software may like to probe the hardware to find out whether misaligned loads and stores are emulated or performed in hardware. That, or code needs to perform a benchmark at start-up...

michaeljclark, 2019-11-08:
This branch updates the array /decoder/ to use aligned loads and stores:

- https://github.com/michaeljclark/vlu/tree/aligned-loads-stores

The /encoder/ will need to be updated for this to be a useful benchmark for misaligned load/stores.

The benchmarks labeled `VLU-raw` don't perform any misaligned loads and stores. It's just the benchmarks labeled `VLU-pack` that use misaligned loads and stores. The benchmarks currently all contain memory accesses. It may be necessary to create benchmarks that do not perform memory accesses, to look at the primitive alone, although it kind of makes sense that these primitives are compared combined with memory bandwidth.

This is a pretty good example of a code that would like to know whether aligned loads and stores are fast or whether to completely avoid them. Currently, it is hard-coded in `vlu.h`.

```
+#define USE_MISALIGNED_LOADS 0
```

michaeljclark, 2019-11-08:
Also note: the pack routines need endian swaps for big-endian architectures, however, it is not a concern at present. It would be relevant if someone wanted to test a big-endian mode. I'm using this code on x86 so it is not currently necessary. Interestingly ASN.1 BER used in X.509 is big-endian, but new ciphers designed this century, such as Salsa20 are little-endian. I'm intending to use VLU in an analog to ASN.1 BER, but it will be optimized for little-endian architectures not big-endian architectures.

michaeljclark, 2019-11-10:
I merged support for aligned accesses but default to unaligned accesses on x86.

cliffordwolf, 2019-11-11:
Thanks!

The following code in your README doesn't look right:

```
struct vlu_result vlu_decode_56c(uint64_t uvlu)
{
    int t1 = __builtin_ctzll(~uvlu);
    bool cont = t1 > 7;
    int shamt = cont ? 8 : t1 + 1;
    uint64_t mask = ~(-!cont << (shamt << 3));
    uint64_t num = (uvlu >> shamt) & mask;
    return (vlu_result) { num, shamt | -(int64_t)cont };
}
```

I think `mask` should be something like the following instead:

```
uint64_t mask = shamt == 8 ? -(uint64_t)1 : ~(-(uint64_t)(!cont) << (shamt << 3));
```

Two changes wrt your version: first, the result of `!cont` is an int, so we need to cast it to `uint64_t` before the prefix minus so that we get a 64-bit bitmask, and second shifting a 64-bit int by 64 (aka `8 << 3`) is undefined, so we need to avoid it and generate the all-bits-set mask explicitly for this case.

At least with GCC targeting RV64 I get incorrect results if I leave out either change.

cliffordwolf, 2019-11-11:
Also, I think this is not quite correct:

```
uint64_t num = (uvlu >> shamt) & mask;
```

I think this should be the following instead:

```
uint64_t num = (uvlu & mask) >> shamt;
```

Otherwise you'll end up with upper bits of the next VLU integer in the result.

cliffordwolf, 2019-11-11:
I have now added examples for unsigned VLU and unsigned LEB128:
https://github.com/riscv/riscv-bitmanip/tree/master/examples/vluleb

Results with the current code:

```
uvlu56_encode_reference   43164 instructions  (48.2 inst / dword)
uvlu56_encode_bitmanip    12551 instructions  (14.0 inst / dword)

uvlu56_decode_reference   51614 instructions  (57.6 inst / dword)
uvlu56_decode_bitmanip    11654 instructions  (13.0 inst / dword)

leb128_encode_reference   36845 instructions  (41.1 inst / dword)
leb128_encode_bitmanip    10759 instructions  (12.0 inst / dword)

leb128_decode_reference   30428 instructions  (34.0 inst / dword)
leb128_decode_bitmanip    11657 instructions  (13.0 inst / dword)
```

Here are my bitmanip implementations. (I left out returning the size of the unencoded/encoded value as it's for the most part merely an exercise in returning multiple values):

```
uint64_t uvlu56_encode_bitmanip(uint64_t num)
{
        long shamt = (_rv64_clz(num) - 70) / -7;
        return ((num << shamt) | _rv64_slo(0, shamt-1)) & _rv64_orc(num);
}

uint64_t uvlu56_decode_bitmanip(uint64_t uvlu)
{
        long shamt = _rv64_ctz(uvlu + 1) + 1;
        return (uvlu >> shamt) & _rv64_slo(0, (shamt << 3)-shamt);
}

uint64_t leb128_encode_bitmanip(uint64_t in)
{
        uint64_t mask = _rv64_orc8(0x80);
        uint64_t out = _rv64_bdep(in, ~mask);
        uint64_t t = (((uint64_t)!out)-1) >> (_rv64_clz(out) & 63);
        return out | (mask & t);
}

uint64_t leb128_decode_bitmanip(uint64_t in)
{
        uint64_t mask = _rv64_orc8(0x80);
        long t = (_rv64_ctz(~_rv64_bext(in, mask)) << 3) + 7;
        return _rv64_bext(in & _rv64_slo(0, t), ~mask);
}
```

closed 2019-11-11 by cliffordwolf

michaeljclark, 2019-11-12:
> Also, I think this is not quite correct:
> 
> ```
> uint64_t num = (uvlu >> shamt) & mask;
> ```
> 
> I think this should be the following instead:
> 
> ```
> uint64_t num = (uvlu & mask) >> shamt;
> ```
> 
> Otherwise you'll end up with upper bits of the next VLU integer in the result.

The shift value is composed using the size of the decoded word after the prefix has been shifted off. I have one test for values that should be masked as an earlier version one of my earlier versions had this issue. I will double-check it and add more.

Thanks for taking the time to look into VLU btw... I want to add 128-bit support too, which is a bit of a pain to get to working across GCC, Clang, and MSVS across multiple architectures. GCC and Clang as you know have __int128 _(1)_ which oddly is using Microsoft's scheme for bit-width sized integer types: __int8, __int16, __int32, __int64 _(2)_.  I can use __m128 across all compilers, but only on Intel _(3)_.

[1] https://quuxplusone.github.io/blog/2019/02/28/is-int128-integral/
[2] https://docs.microsoft.com/en-us/cpp/cpp/int8-int16-int32-int64?view=vs-2019
[3] https://github.com/llvm-mirror/clang/blob/master/lib/Headers/xmmintrin.h

BTW - Wouldn't it be nice if folk juse used u8, u18, u32, u64, etc, but ya, what do we do for signed? i8, i16, i32, i64 or s8, s16, s32, s64? to me _i(n)_ seems size agnostic. Maybe it really means (signed) integer?

cliffordwolf, 2019-11-13:
> The shift value is composed using the size of the decoded word after the prefix has been shifted off.

I assume you mean mask value, not shift value?

```
int t1 = __builtin_ctzll(~uvlu);
bool cont = t1 > 7;
int shamt = cont ? 8 : t1 + 1;
uint64_t mask = ~(-!cont << (shamt << 3));
```

`shamt` is the number of trailing one bits, plus one, which is the size of the word _before_ decode in bytes. So `shamt << 3` is the number of bits in the word before decode. The size after decode would be `(shamt << 3) - shamt` afaict.

michaeljclark, 2019-11-17:
Yes, I meant mask value (I am having concentration issues due to epilepsy meds ... for pain).

The *mask* value is composed using the decoded size after the prefix has been shifted off. There was a bug in the README which I have just fixed and pushed. The code in `vlu.h` was correct. However, it would be possible to mask before shifting off the prefix. I am not sure what is best.

```
diff --git a/README.md b/README.md
index 622ed9a46aa1..ad70c6fd2293 100644
--- a/README.md
+++ b/README.md
@@ -174,7 +174,7 @@ struct vlu_result vlu_decode_56c(uint64_t uvlu)
     int t1 = __builtin_ctzll(~uvlu);
     bool cont = t1 > 7;
     int shamt = cont ? 8 : t1 + 1;
-    uint64_t mask = ~(-!cont << (shamt << 3));
+    uint64_t mask = ~(-(int64_t)!cont << (shamt * 7));
     uint64_t num = (uvlu >> shamt) & mask;
     return (vlu_result) { num, shamt | -(int64_t)cont };
 }
```

Thanks for pointing this out...