riscv/riscv-bitmanip #53: maybe use pcnt(rs1 ^ rs2) ?

brucehoult, 2019-11-30:
In https://groups.google.com/d/msg/comp.arch/8MR8_O-wCeE/8pyiGYz8AQAJ Pedro Pereira suggests:

In the latest bitmanip extension document, the popcount opcode is defined as:

  rd = pcnt(rs)

a more useful primitive would be:

  rd = pcnt(rs1 ^ rs2)

Since the RISC-V has a zero register (x0), the suggested version could
encode the first one as "rd = pcnt(rs ^ x0)".

I don't imagine that reading one extra register and
performing a xor would make the instruction need more cycles.

brucehoult, 2019-11-30:
Certainly having Hamming distance available directly would slightly improve many algorithms, and the cost is low (using a binary op instead of unary).

montedalrymple, 2020-03-13:
I believe it would be more useful if the primitive were:

rd = pcnt(rs1 & ~rs2)

This allows for pcnt operations on bytes, halfwords or fields in words. This makes the instruction
useful in dealing with sub-32-bit applications or packed-data applications.

An immediate version will also be useful, for the same reason. The ~imm takes advantage of
the sign-extend nature of immediate operands to make a direct pcnt.b 

rd = pcnt(rs1 & ~imm)

brucehoult, 2020-03-14:
"rd = pcnt(rs1 & ~rs2)" seems like a possibility. As with the xor variant it allows using x0 as rs2 for a simple popcnt.

An immediate version is I think out of the question. It would be used in far too little code to justify the opcode space.

kdockser, 2021-04-06:
Population count, believe it or not, is not a trivial instruction to implement in terms of levels of logic. To help reduce timing pressure, we have kept it as its own instructions. Of course, implementations are free to fuse XOR or ANDN with Population Count.

closed 2021-04-06 by kdockser