riscv/riscv-bitmanip #56: Exceptions

Imperas, 2020-01-22:
@cliffordwolf 
Hi Clifford
for an XLEN=32 or XLEN=64 Implementation, should the following instruction raise an exception ?

sbseti x1, x2, 127

Thx
Lee

cliffordwolf, 2020-02-14:
Yes. The as for `slli x1, x2, 127`. (In both cases, `sbseti` and `slli`, binutils would not allow you to actually create that instruction word tho.)

closed 2020-02-14 by cliffordwolf

pdonahue-ventana, 2020-02-18:
There is no requirement that reserved instructions raise exceptions.  The base architecture says:
> The behavior upon decoding a reserved instruction is UNSPECIFIED.

Claire: Did you simply mean that the instruction is reserved or literally that it should raise an exception?

eroom1966, 2020-02-18:
> For RV32I, SLLI, SRLI, and SRAI generate an illegal instruction exception if imm[5] ̸= 0

Doesn’t it fall into the same category as above ?

pdonahue-ventana, 2020-02-18:
> > For RV32I, SLLI, SRLI, and SRAI generate an illegal instruction exception if imm[5] ̸= 0
> 
> Doesn’t it fall into the same category as above ?

What are you quoting?  I don't see that text in bitmanip, the unprivileged spec, or the privileged spec.  All I see is this in section 5.2 of the unprivileged spec:
> SLLIW, SRLIW, and SRAIW encodings with imm[5] ̸= 0 are reserved.

eroom1966, 2020-02-18:
User level ISA 2.2
https://content.riscv.org/wp-content/uploads/2017/05/riscv-spec-v2.2.pdf

pdonahue-ventana, 2020-02-18:
Looking through the history, that changed in August 2018 from what you quoted to what I quoted.

A newer version (dated 2019/12/13) is at https://riscv.org/specifications/isa-spec-pdf/ and, of course, there have been over a dozen new draft versions on github in the 2 months since that December version.

Imperas, 2020-02-19:
@cliffordwolf @kdockser 
I think somebody probably needs to refer this for agreement
Should this instruction generate an illegal exception ?

Imperas, 2020-02-19:
I note the following text, in the ratified 2.2 user spec page 5

> Reserved encodings are currently not defined but are saved for future standard extensions

In which case an SRLI for RV64 **is** defined as part of the RV64I standard extension, and is therefore **not** reserved.
So trying to execute SRLI(RV64) on an RV32 machine, would be an illegal instruction.
If I understand correctly, the purpose of defining RESERVED, is to protect against future definition as part of a standard extension

It would be good to get clarification on this from the author
@aswaterman