riscv/riscv-bitmanip #57: Missing library file "vsupport.h"

federicorossifr, 2020-02-03:
I'm trying to compile examples with the compiled toolchain from the riscv repository. However, I could not find the said file in any repository here. From where does it come from?

jim-wilson, 2020-02-03:
It is the riscv-ovpsim vector examples that use vsupport.h, and there is no vector support on the bitmanip branch.  It is a separate unrelated draft extension.  Just ignore the vector stuff.  There isn't any useful compiler support for it yet.

federicorossifr, 2020-02-04:
Thank you for your reply.

closed 2020-02-04 by federicorossifr

dovedevic, 2020-08-17:
I'm looking to benchmark some vector instructions with an application in mind. Is this still the case? Currently using the `riscv64-unknown-elf-gcc` compiler. Any chance at understanding what `enableVEC()` does explicitly, as well as the types `v512P, v512T, VEC32, and Uns32`?

jim-wilson, 2020-08-24:
This is still not the vector extension.  The imperas simulator apparently has some vector support, but you would have to ask them.

LarryImperas, 2020-08-24:
Readme for riscvOVPsim shows full support for the vector instructions:  https://github.com/riscv/riscv-compliance/blob/master/riscv-ovpsim/README.md.  riscvOVPsim can be used as an ISS, however, it has some limitations in functionality.  If you run into those, try the full OVPsim simulator from the Open Virtual Platforms website.  You can start here:  https://www.ovpworld.org/info_riscv.