riscv/riscv-bitmanip #63: internal compiler error: in decompose, at rtl.h:2279

aefimov-cb, 2020-05-29:
Hello,

The compilation of source foo.c (see below) fails with internal error when the compiler is run as follows:
```
$ riscv64-unknown-elf-gcc -O2 -march=rv64ib -mabi=lp64 -S -o foo.S foo.c
during RTL pass: combine
foo.c: In function ‘foo’:
foo.c:4:1: internal compiler error: in decompose, at rtl.h:2279
```
Source code foo.c:
```
int foo(int n)
{
    return n + 0x7fffffff;
}
```
Additional notes:
* Replacing -O2 with -O{1,0,s} does the same. When -O is omitted, the program compiles.
* When -march=rv64ib is replaced with -march=rv64i, the program compiles.
* When -march=rv64ib -mabi=lp64 is replaced with -march=rv32ib -mabi=ilp32, the program compiles.

gcc -v log:
```
[user@s01 bug]$ ../riscv64b/bin/riscv64-unknown-elf-gcc -Os -march=rv64ib -mabi=lp64 -v -S -o foo.S foo.c
Using built-in specs.
COLLECT_GCC=../riscv64b/bin/riscv64-unknown-elf-gcc
Target: riscv64-unknown-elf
Configured with: ../riscv-gcc/configure --prefix=/home/user/riscv-bitmanip/riscv64b --target=riscv64-unknown-elf --enable-languages=c --disable-libssp
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 10.0.0 20190929 (experimental) (GCC) 
COLLECT_GCC_OPTIONS='-Os' '-march=rv64ib' '-mabi=lp64' '-v' '-S' '-o' 'foo.S'
 /home/user/riscv-bitmanip/riscv64b/libexec/gcc/riscv64-unknown-elf/10.0.0/cc1 -quiet -v foo.c -quiet -dumpbase foo.c -march=rv64ib -mabi=lp64 -auxbase-strip foo.S -Os -version -o foo.S
GNU C17 (GCC) version 10.0.0 20190929 (experimental) (riscv64-unknown-elf)
        compiled by GNU C version 4.8.5 20150623 (Red Hat 4.8.5-39), GMP version 6.0.0, MPFR version 3.1.1, MPC version 1.0.1, isl version none
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
ignoring nonexistent directory "/home/user/riscv-bitmanip/riscv64b/lib/gcc/riscv64-unknown-elf/10.0.0/../../../../riscv64-unknown-elf/sys-include"
#include "..." search starts here:
#include <...> search starts here:
 /home/user/riscv-bitmanip/riscv64b/lib/gcc/riscv64-unknown-elf/10.0.0/include
 /home/user/riscv-bitmanip/riscv64b/lib/gcc/riscv64-unknown-elf/10.0.0/include-fixed
 /home/user/riscv-bitmanip/riscv64b/lib/gcc/riscv64-unknown-elf/10.0.0/../../../../riscv64-unknown-elf/include
End of search list.
GNU C17 (GCC) version 10.0.0 20190929 (experimental) (riscv64-unknown-elf)
        compiled by GNU C version 4.8.5 20150623 (Red Hat 4.8.5-39), GMP version 6.0.0, MPFR version 3.1.1, MPC version 1.0.1, isl version none
GGC heuristics: --param ggc-min-expand=30 --param ggc-min-heapsize=4096
Compiler executable checksum: bff2308ac495c30be0d25ad6caff4627
during RTL pass: combine
foo.c: In function ‘foo’:
foo.c:4:1: internal compiler error: in decompose, at rtl.h:2279
    4 | }
      | ^
0x556a3e wi::int_traits<std::pair<rtx_def*, machine_mode> >::decompose(long*, unsigned int, std::pair<rtx_def*, machine_mode> const&)
        ../../riscv-gcc/gcc/rtl.h:2277
0xbd4961 wi::int_traits<std::pair<rtx_def*, machine_mode> >::decompose(long*, unsigned int, std::pair<rtx_def*, machine_mode> const&)
        ../../riscv-gcc/gcc/wide-int.h:3102
0xbd4961 wide_int_ref_storage<std::pair<rtx_def*, machine_mode> >
        ../../riscv-gcc/gcc/wide-int.h:1032
0xbd4961 generic_wide_int<std::pair<rtx_def*, machine_mode> >
        ../../riscv-gcc/gcc/wide-int.h:790
0xbd4961 add<std::pair<rtx_def*, machine_mode>, std::pair<rtx_def*, machine_mode> >
        ../../riscv-gcc/gcc/wide-int.h:2422
0xbd4961 simplify_const_binary_operation(rtx_code, machine_mode, rtx_def*, rtx_def*)
        ../../riscv-gcc/gcc/simplify-rtx.c:4318
0xbd9cde simplify_binary_operation(rtx_code, machine_mode, rtx_def*, rtx_def*)
        ../../riscv-gcc/gcc/simplify-rtx.c:2156
0x1227a71 combine_simplify_rtx
        ../../riscv-gcc/gcc/combine.c:5804
0x122a492 subst
        ../../riscv-gcc/gcc/combine.c:5726
0x122a108 subst
        ../../riscv-gcc/gcc/combine.c:5667
0x122c4ed try_combine
        ../../riscv-gcc/gcc/combine.c:3422
0x12323d8 combine_instructions
        ../../riscv-gcc/gcc/combine.c:1305
0x12323d8 rest_of_handle_combine
        ../../riscv-gcc/gcc/combine.c:15066
0x12323d8 execute
        ../../riscv-gcc/gcc/combine.c:15111
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
```

System info: CentOS 7, uname -a: Linux XXXXXXXXX 3.10.0-514.el7.x86_64 #1 SMP Tue Nov 22 16:42:41 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

jim-wilson, 2020-05-29:
The problem is the use of sbseti to load constants.  GCC is trying to split the constant 0x7fffffff into 0x80000000 and -1, and use sbseti to load the 0x80000000.  However, this is a 32-bit constant we are loading, and a 32-bit 0x80000000 is not a single bit constant as it needs to be sign-extended, so we can't use sbseti to load it.  We can load a 64-bit 0x80000000 with sbseti, but unfortunately we don't know the final mode here.  So we need to refuse to use sbseti always for 0x80000000.  sbseti is undesirable for any bit in the lower 32-bits, as we should use addi or lui to load that.  So the fix is to only allow sbseti to be used for bits in the upper 32-bits.

gcc patch submitted as
https://github.com/riscv/riscv-gcc/pull/185

My testcase that I used to test my patch.

/* Should not crash.  Should not generate an sbseti.  */
int
no_sbseti_1 (int n)
{
  return n + 0x7fffffff;
}

/* Should not generate an sbseti.  */
int
no_sbseti_2 (void)
{
  return 0x40000000;
}

/* Should generate an sbseti.  */
long
sbseti_3 (void)
{
  return 0x0000000100000000;
}

/* Should not generate an sbseti.  */
int
no_sbseti_4 (void)
{
  return 0x80000000;
}

/* ??? Could generate an sbseti, but not yet supported.  */
long
unimp_sbseti_5 (void)
{
  return 0x0000000080000000;
}

aefimov-cb, 2020-06-01:
Hi Jim,

Thanks for the patch.
I was able to build my workload with your patch, so I think the issue is resolved. Should I close the issue or is it better to wait until the patch is accepted?

Regards,
Andrey

jim-wilson, 2020-06-02:
Thanks for the confirmation.  bitmanip isn't really my project, so I wanted to wait at least one work day for comment before committing my riscv-gcc patch.  I did that today.  I don't have write permissions for this project, so I am not able to close this issue myself.

closed 2020-06-05 by aefimov-cb