riscv/riscv-bitmanip #66: Incorrect operands type for  bswapdi2 in gcc/config/riscv/bitmanip.md

aefimov-cb, 2020-06-02:
Hello,

Currently bswapdi2 is defined in bitmanip.md as follows:
```
(define_insn "bswapdi2"
  [(set (match_operand:SI 0 "register_operand" "=r")
	(bswap:SI (match_operand:SI 1 "register_operand" "r")))]
  "TARGET_64BIT && TARGET_BITMANIP"
  "grevi\t%0,%1,0x38"
  [(set_attr "type" "bitmanip")])
```
Looks like SI should be replaced with DI in this definition.

My test was a function listed below (originally defined in libgcc/libgcc2.c):
```
typedef long DItype;

DItype
__bswapdi2 (DItype u)
{
  return ((((u) & 0xff00000000000000ull) >> 56)
          | (((u) & 0x00ff000000000000ull) >> 40)
          | (((u) & 0x0000ff0000000000ull) >> 24)
          | (((u) & 0x000000ff00000000ull) >>  8)
          | (((u) & 0x00000000ff000000ull) <<  8)
          | (((u) & 0x0000000000ff0000ull) << 24)
          | (((u) & 0x000000000000ff00ull) << 40)
          | (((u) & 0x00000000000000ffull) << 56));
}
```
Before replacement of SI with DI the function assembly was broken:
```
__bswapdi2:
        addi    sp,sp,-16
        sd      ra,8(sp)
        call    __bswapdi2    /// <-- infinite recursion
        ld      ra,8(sp)
        addi    sp,sp,16
        jr      ra
```
After replacement it looks good:
```
__bswapdi2:
        grevi   a0,a0,0x38
        ret
```
Please, make a fix.

jim-wilson, 2020-06-03:
Looks like a simple copy paste error from the bswapsi2 pattern.  I add riscv-gcc pr #186 to fix.

jim-wilson, 2020-06-03:
The riscv-gcc patch has been committed.

aefimov-cb, 2020-06-05:
Thanks for the fix.
I'm closing the issue.

closed 2020-06-05 by aefimov-cb