riscv/riscv-bitmanip #72: The bits of imm field

davidmlw, 2020-06-16:
https://github.com/riscv/riscv-bitmanip/blob/a05231da8eba98dce24d7dec3fd64724cc8d7088/texsrc/bext.tex#L1867-L1879

`imm` field, needs 5 bits in RV32, 6 bits in RV64, while here in spec, it reserves 7 bits. The definition of `SLLI` in riscv-spec is two-fold for RV32 and RV64. It could be better to define accordingly.

jim-wilson, 2020-06-16:
rv128 uses 7 bits for shift counts.

davidmlw, 2020-06-17:
I see. But still it is better to define in rv32, bits 26..25 is `00` or `??`, and in rv64, bit 26 is `0` or `?`.

JamesKenneyImperas, 2020-06-17:
I completely agree that clarification is needed. Elsewhere in the architecture, non-zero values in the cases stated would cause Illegal Instructions, but it is not clearly stated that this is the case in this specification.

kdockser, 2021-04-06:
This is being clarified in the AsciiDoc version of the Zb[abcs] spec that will soon (hopefully) be available for Public Review. 
The plan for all of the ISA is to have just one central repository that holds the official encodings: https://github.com/riscv/riscv-opcodes
The base ISA takes its encodings directly from here. As all specs migrate to AsciiDoc, we will be moving to a standard template for instruction descriptions where the encoding is extracted from this repository.

closed 2021-04-06 by kdockser