riscv/riscv-bitmanip #74: Removal of addwu/subwu/addiwu instructions

clairexen, 2020-06-23:
We are considering removal of addwu/subwu/addiwu from the bitmanip extension. They are very cheap to implement, but there are concerns if they add enough to justify even the small implementation cost and/or ISA encoding space cost.

Please post relevant code examples and benchmarks below.

(Note that all *wu and *u.w instructions can only replace an add/sub instruction and a pack instruction. So they never replace more than two instructions, thus only relatively common code patterns are of interest.)

clairexen, 2020-06-23:
If we remove `addiwu` then it won't be possible anymore to load a zero-extended 32-bit value into a register in only two instructions with `lui+addiwu`. (Note that `addi+pack` can handle the cases that `lui+addiwu` can't handle.)

kasanovic, 2020-06-24:
There might not be a need to materialize a 32-bit zero-extended constant if it is only consumed by prefix-zeroing instructions including stores.

clairexen, 2020-06-24:
I would _guess_ there are many cases where they are consumed by logic, arithmetic, or bitmanip instructions. This needs the be benchmarked ofc. But unfortunately that's hard with current https://github.com/riscv/riscv-gcc/tree/riscv-bitmanip. For example:

```
unsigned long foo() { return 0x00000000fedcba98UL; }
unsigned long bar() { return 0x00000000fffffabcUL; }
```

Produces (irrelevant lines removed):

```
$ /opt/riscv64b/bin/riscv64-unknown-elf-gcc -o - -S -Os -march=rv64ifdb demo.c 
foo:
	li	a0,1068969984
	slli	a0,a0,2
	addi	a0,a0,-1384
	ret
bar:
	addiwu	a0,zero,-1348
	ret
```

`foo()` should be `lui+addiwu`, but instead it's a three instruction sequence (1068969984 = 0x3fb73000).

`bar()` can be implemented in a single `addiwu` and gcc does that correctly.

Looking at this I think benchmarks with current gcc under-count cases where `addiwu` could be useful to materialize constants. (This should be easy enough to fix tho ofc.)

pdonahue-ventana, 2021-02-12:
Was this completed by https://github.com/riscv/riscv-bitmanip/commit/ba8677c2d9b2b35edff38c0f1e78f35c08e873d0?

Because ADDIWU is still in the encoding tables, it took me quite a while to figure out what was going on with that instruction.  The minus sign and the sentence at the very bottom of page 45 were easy to miss.  Can this encoding changes table be put in its own section 2.12.1 or in some other way clearly distinguished from the rest of the tables?

pdonahue-ventana, 2021-02-12:
And was https://github.com/riscv/riscv-bitmanip/commit/ba8677c2d9b2b35edff38c0f1e78f35c08e873d0 merged directly to master without a PR?  Because this issue was never closed and there was no PR, people watching this repo didn't get a notification of the change.

ptomsich, 2021-06-22:
Rolled up into the public review release 1.0.0.

closed 2021-06-22 by ptomsich