riscv/riscv-bitmanip #82: Encoding for xperm instructions

clairexen, 2020-08-20:
the encoding for the xperm instructions is still missing

JamesKenneyImperas, 2020-10-02:
Is there a timeframe for resolution of this issue? I've implemented the K scalar extension in riscvOVPsim and would like this shared instruction for completeness.

Thanks,

James.

ben-marshall, 2020-10-02:
For those that the news hasn't filtered out too yet - K is the designation for the cryptography extension...

JamesKenneyImperas, 2020-10-02:
Oops, sorry, too cryptic of me...

JamesKenneyImperas, 2021-01-21:
I think this issue is now resolved and can be closed, isn't it?

kdockser, 2021-01-21:
> CAUTION: This email originated from outside of the organization.

There is a proposed encoding for Xperm – but it hasn’t been approved yet.
Since this and other BitManip instructions that were not part of the approved set of Bz[a,b,c,s], I am spawning another Opcode and Consistency Review for the BitManip instructions required by Crypto Scalar.

Cheers,
Ken