riscv/riscv-bitmanip #83: Get rid of subu.w, swap args on addu.w

clairexen, 2020-08-26:
There's the proposal to remove subu.w and change addu.w so that rs1 is the argument being zero-extended instead of rs2. This way the addu.w argument semantic would match sh?addu.w and slliu.w.

jim-wilson, 2020-08-26:
Someone should do a study to see how this affects code size and performance.  Preferably by modifying a compiler and comparing the results with and without the change.

Rolfy360, 2020-10-24:
Suggest that addu.w can be changed to addu.wx
With this format,
“addu.wx rd, rs1, rs2” means rs1 is word-size and rs2 is xlen.
Borrow this idea from vector extension.

pdonahue-ventana, 2021-02-12:
@aswaterman: I think that you addressed this so should it be closed?

aswaterman, 2021-02-12:
Yes, closed by #89

closed 2021-02-12 by aswaterman