riscv/riscv-bitmanip #84: Status rev, orc16, rev.w, orc16.w in Zbb

clairexen, 2020-08-26:
There have been discussions about removing orc16 from Zbb and rev.w, orc16.w from RV64_Zbb.

There was also a mention of removing rev from Zbb.

jim-wilson, 2020-08-26:
If you have clz but not ctz, then you can implement ctz with rev;clz.  Or vice versa.  This plus intrinsics is the only use in the arm port for the rbit instruction which is roughly same as rev.  But we have both clz and ctz in bitmanip so we don't need rev for this.  I've seen hand written arm code using rbit, but it is uncommon and unlikely to be missed.

rev8 is byte swapping which is very important in some applications.

I don't think that orc* is useful to the compiler, but I haven't tried to use it.

ptomsich, 2021-06-22:
Resolved as part of the 1.0.0 public review draft.

closed 2021-06-22 by ptomsich