riscv/riscv-bitmanip #90: shadd consistency

jnk0le, 2020-10-24:
I see that there is `sh4add` provided in Fig. 2.5, but only up to sh3add was defined.
In my understanding  it's state is currently a bit inconsistent.

- If we want to provide scaling up to XLEN sized elemets, then RV32 should not get `SH3ADD`.
- If we want to provide scaling up to FLEN (even when not implemented), then RV64 should get `SH4ADD`.
- If we want to provide scaling up to implemented FLEN, then this is a cancer and RV32IB should not get `SH3ADD` anyway.
- If we want to provide one set of instructions for RV32, RV64 and RV128, then RV32 and RV64 should get `SH4ADD`. Otherwise RV128 will end up without ability to scale native sized elements.

jnk0le, 2021-01-12:
response from AW:
>Apparently, but a non-resolution to #90 is an appropriate response. Providing shifts up to 3 is about circuit delay; 3 is similar to 2, but 4 adds another level of logic. And supporting shifts by 3 is still plenty useful in RV32 implementations. While I would argue against providing shadd4, it can always be provided as part of a future extension; that decision does not need to be made now.

leaving issue for future extensions to zba extension

kdockser, 2021-04-06:
Closing as there is no immediate plan to add an future extension to the zba sub-extension. We still have plenty of other bitmanip sub-extensions to get to pubic review.
A new issue can always be created if/when we consider an extension to zba.

closed 2021-04-06 by kdockser