riscv/riscv-bitmanip #93: MAX/MINU permuted in documentation vs. binutils

rdolbeau, 2020-11-03:
From the binutils `include/opcode/riscv-opc.h`
```
#define MATCH_MIN 0xa004033
#define MATCH_MAX 0xa005033
#define MATCH_MINU 0xa006033
#define MATCH_MAXU 0xa007033
```
From the riscv-bitmanip documentation `texsrc/bext.tex`
```
|  0000101    |   rs2   |   rs1   | 100 |    rd   |   0110011   |  MIN
|  0000101    |   rs2   |   rs1   | 101 |    rd   |   0110011   |  MINU
|  0000101    |   rs2   |   rs1   | 110 |    rd   |   0110011   |  MAX
|  0000101    |   rs2   |   rs1   | 111 |    rd   |   0110011   |  MAXU
```
The opcode from MINU and MAX are permuted between the two (noticed in https://github.com/SpinalHDL/VexRiscv/pull/148#issuecomment-721050796).

jim-wilson, 2020-11-03:
The encodings for min/max changed recently, and no one is actively maintaining this binutils port.  There are also multiple insns in the process of being added and removed.  There are actually multiple parties working on multiple related gnu toolchain ports, some in public trees and some in private trees.  This is very badly screwed up and not clear when this will be fixed.  Meanwhile, you will have to be patient.

rdolbeau, 2020-11-03:
No problem, it's an easy fix in the binutils tree. Now it's documented for those with similar issues :-)

jim-wilson, 2021-02-24:
This was fixed by riscv-binutils-gdb pull request #241.  So is fixed on riscv-binutils-2.35-rvb branch.

rdolbeau, 2021-02-24:
@jim-wilson Unfortunately when using this repository, the scripts in `tools` points to an older version (https://github.com/riscv/riscv-bitmanip/blob/master/tools/checkout.sh still checkout `riscv-bitmanip` for riscv-binutils-gdb...) so the bug still appear to be present.

So now the bug really is, "riscv-bitmanip repo uses an obsolete toolchain", I suppose.

brson, 2021-02-28:
I tried to cherry-pick https://github.com/riscv/riscv-binutils-gdb/pull/241 into the riscv-binutils-gdb riscv-bitmanip branch, so that the checkout.sh script would work again, but it didn't result in a working build. The riscv-bitmanip branch is so old that it seems like the checkout script might need to be updated to use more recent upstream branches.