riscv/riscv-bitmanip #97: some zbs instruction encodings are redundant with i

jim-wilson, 2020-11-25:
sbexti a0,a0,0 is identical in effect to andi a0,a0,1.  Since andi is compressable and sbexti is not, a compiler should always emit andi instead to avoid a code size regression, and the sbseti 0 encoding could be reserved.

sbseti a0,a0,0 ... sbseti a0,a0,10 are identical to ori a0,a0,0x1 ... ori a0,a0,0x400, and similarly sbinvi 0 to 10 are identical to xori 0x1 to 0x400.  ori and xori are not compressable so there is no code size regression issue here, but the duplication is unnecessary, and these could be reserved encodings.

There is no andni, so there is no redundant encoding issue with sbclri.

There is no oriw or xoriw, so there are no redundant encodings with the sb*iw instructions.

RogierBrussee, 2020-12-03:
The same redundant encoding issue is there for sbclri because it is identical with andi ~0x1 ..... ~0x400. 
Moreover there is a codegen regression issue for bit 0..3 because andi can be compressed.

RogierBrussee, 2020-12-04:
Come to think of it 

sbgeti rd rs1 XLEN-1 is identical in effect to slti rd rs1 0. 

sb{set/clr/inv}wi rd rs1 imm  is equivalent to sb{set/clr/inv}i rd rs1 imm for imm < 31,   (but sb{set/clr/inv}wi rd 31 corresponds to   sb{set/clr/inv}i rd 31; sext.w rd rd )

ptomsich, 2021-06-22:
The suggestion to make those encodings that shadow instructions from RV32I/RV64I as RESERVED/RFU did not find a consensus (during the earlier discussions to finalize 1.0.0) based on the added decoding complexity.

closed 2021-06-22 by ptomsich