riscv/riscv-bitmanip #98: Some instructions are orphaned (belong to no extension)

rdolbeau, 2020-12-07:
Hello,

In the latest draft (as of 2020/12/07), some instructions are defined but do not belong to any extension in the table 2.1 on page 4.

Specifically, `slo`/`sro` and variants are missing - yet are still defined in section 2.1.8. In draft 0.92 they belong to Zbb itself.

Are those instructions removed, or is that just an oversight ?

Cordially,

kdockser, 2020-12-07:
> CAUTION: This email originated from outside of the organization.

Hi Romain,

The slo/sro  instructions were intentionally removed from the sub-extensions (Zba, b, c and s) currently being brough through the “Opcode and Consistency Review” on the way to specification Freeze.

Do you see a particular need for these instructions or did you just happen to notice that they are no longer instead in any of the sub-extensions?

Thanks,
Ken

rdolbeau, 2020-12-07:
Hello,

No specific need, just noticed they were gone while updating the definition file for my code generator.

There's other defined subsets not in your list (p, e, f, t, r, m), it simply felt strange for those instructions to not be anywhere.

Cordially,

kdockser, 2020-12-07:
As I recall, the thinking was there was not much value to these instructions, so they were not moved to any of the other sub-extensions. This decision was supported by the gcc compiler not picking them up on SPECINT. However, should a compelling use for these instructions be brought forward, we could include them in one of the subsequent sub extensions.

Thanks,
Ken

rdolbeau, 2020-12-07:
OK thank you, not an issue then.

closed 2020-12-07 by rdolbeau