riscv/riscv-bitmanip #108: Stale implementation note for SLO/SRO

topperc, 2021-01-24:
The implementation note for SLO/SRO says "When implementing this circuit, the only change in the ALU over a standard logical shift is that the value shifted in is not zero, but is a 1-bit register value that has been forwarded from the high bit of the instruction decode".

It looks like SLO and SLL differ in bit 29 not the high bit. Same for SRO/SRL. Is this implementation note referring to an earlier encoding for SLO/SRO?

David-Horner, 2021-01-24:
Yes, you are correct, it previously read:

> When implementing this circuit, the only change in the ALU over a
> standard logical shift is that the value shifted in is not zero, but is
> a 1-bit register value that has been forwarded from **the high bit of the
> instruction decode**. This creates the desired behavior on both logical
> zero-shifts and logical ones-shifts.

And true, the opcodes differ in a single bit and it is not the "sign-bit" in the instruction.

It is the operation, however, that uses the sign-bit of the rs2 operand (or the immediate value for SLOI/SROI).
The implementation note should be referring to that sign bit; for SLOI/SROI it is technically correct but confusing. 

Thus it should probably read:

>"When implementing this circuit, the only change in the ALU over a standard logical shift is that the value shifted in is not zero, but is a 1-bit register value that has been forwarded from the **high bit of the second operand**".

There might be reason to added a comment that, for the immediate variant the instruction decode fully defines the shift in value.
But I think that too obvious to include. Even though it appears that was the reason of the original text; noting that immediate sign matched with register high bit in RV32 and thus could reduce decode complexity.

topperc, 2021-01-24:
The high bit of the second operand shouldn't be relevant here. The SLO/SRO instructions shift in ones while the standard logical shift instructions SLL/SRL shifts in zeros. So I think the implementation note was trying to say that whether you shift in ones or zeroes can be determined by a specific bit in the opcode. That bit used to be the sign bit, but now appears to be bit 29.

David-Horner, 2021-01-24:
You are absolutely correct. (of course you are).

So, the question is if this should read:
>"When implementing this circuit, the only change in the ALU, that also supports the standard logical shift, is that the value shifted in is not zero, but is a 1-bit register value forwarded from the bit 29 of the instruction."

or something similar, or not have the comment at all.

kdockser, 2021-01-28:
Since these shift-in-zero instruction do not have an official encoding right now, the question is moot. Furthermore, the shift-in-zero instructions are not currently part of any proposed sub-extensions. They are being added to a new section of the specification for instructions that have been considered but not approved. Such instructions could be approved in the future should sufficient information be provided to prove their utility and value.

closed 2021-01-28 by kdockser