riscv/riscv-bitmanip #110: [non-]destructive ternary instructions (B vs. P)

rdolbeau, 2021-02-09:
Hello,

Following #76, ternary are now in there own subextension (Zbt). They use bit 31-27 (upper 5 bits of func7, rs3) as the index to the third input register - they are true non-destructive ternary operations, but consumes a lot of opcode space.

Meanwhile, [P](https://raw.githubusercontent.com/riscv/riscv-p-spec) (packed simd) has some destructive ternary operations (e.g. pbsada, ...) where the 'virtual' third input operand is using rd as the index. So to implement both, an implementation need to look for the third input in two different positions depending on the instruction/extension...

Maybe B could reduce opcode space usage by defining destructive operations as well ? Of course it depends how often the 'third' input is reused, but the order/semantic could be changed to make sure the least-reused operand is sourced from rd.

Of course an alternative solution would be for P to use the same non-destructive encoding scheme as B, but operations like pbsada are very likely to be used in 'destructive' mode so it would probably not be a good use of opcode space.

kdockser, 2021-02-12:
> CAUTION: This email originated from outside of the organization.

Hi Romain,

There are currently no ternary instructions in the base architecture. As I recall, the only ratified extensions that include these are the floating point extensions where they are used for the multiply add instructions.

Ternary instructions can be powerful, but they do add additional overhead to a design in two main areas:

  1.  Opcode space for constructive (i.e., non-destructive) operations where three source operands and a destination must be specified.
  2.  A third read port for operands

I agree with you that the first area can be resolved by defining these instructions as being destructive and using the destination field as the specifier of one of the sources. This has been discussed and considered for vector as well as future floating-point extensions.

The impact of the second area can be mitigated by implementations expanding such instructions into two internal instructions. However, this adds complexity to simpler/smaller designs – the very people who don’t want the cost of the ternary instructions in the first place.

You also make a good point about how we need to sync up with the P extension where there is overlap and work towards a common instructions with common encodings. We had started this last year and it is on my to do list to resume these discussions.

While the ternary instructions are currently in the BitManip proposal spec, there is no immediate plan to bring them forward towards ratification. There is a lot of interest in BitManip as well as a lot of pushback from implementers who don’t want to have to add a 3rd read port which will only be used for a very limited set of instructions. Perhaps in the future there will be enough high-value ternary instructions proposed that the third read port would become acceptable to implementers.

Thanks,
Ken

jnk0le, 2021-02-13:
Since P extension has a high chance to co-exist with zfinx and it got assigned separate major opcode, maybe they could fit their ternary/fma opcodes in R4 type instructions as well?

clairexen, 2021-04-01:
Where would you put the immediate versions of the shift-like ternary instructions? ideally each one of those should provide up-to 7 immediate bits to accommodate RV128, i.e. 4x the size of a ternary instruction getting it's 3rd argument from a register. the proposal I shared on the list defines 4 such instructions FSLI, FSRI, SAPI, CUTI. If we would put them there as well, then they alone would already take up half of that encoding space. If we put them somewhere else then your proposed solution only provides about one quarter of the encoding space we'd need for FSL, FSR, SAP, CUT, MIX, MUX, FSLI, FSRI, SAPI, CUTI.

Edit: Obviously FSLI and FSRI are redundant. :facepalm: I'll update my proposal to only contain FSRI..