riscv/riscv-bitmanip #111: no Zbb opcode for zext.h

cincocerros, 2021-02-14:
in 0.93, zext.h is shown as instruction in the Zbb group, but there's no opcode for it. The text shows it can be a pseudo-instruction using pack, but pack isn't in the Zbb group, so if one only implements the Zbb group, what to do?

jim-wilson, 2021-02-14:
zext.h is a subset of pack.  If you only implement Zbb, then you only implement the subset of pack that maps to zext.h.

cincocerros, 2021-02-14:
I think that approach is wrong. zext,h is trivial to do directly in hardware (in fact i'm pretty sure the mux to do so is in our hdw somewhere) so why make someone implement a more complicated thing (pack subset) to do it. If it is important to have this as an instruction (questionable), then give it an opcode, otherwise make it purely software. For me, the mixing of what's done in software (pseudo instructions) and what's to be done in hardware is confusing in the document. I thought this was supposed to be the instruction set of a processor, ie hardware.
Anyway, not a big deal, I'll live with it.

closed 2021-02-14 by cincocerros

jim-wilson, 2021-02-14:
The document doesn't specify how you implement the zext.h subset of pack in hardware.  If you want to implement it directly as a zext.h you are welcome to do so, and in fact that is the sensible approach.  The document only specifies that the encoding of zext.h is the same as a subset of encodings of pack.  That avoids wasting the finite opcode encoding space by having two encodings that map to the same operation.

cincocerros, 2021-02-14:
ok, i wasn't thinking clearly, i've got it now, thanks