riscv/riscv-bitmanip #112: move Zbt into FMA4 encoding

jnk0le, 2021-02-14:
related to #110 as there is push for destructive ternary instructions.

There are 2 reserved RM encodings in funct 3, two extra bits from fmt, times 4 FMA4 major opcodes. Totaling at 32 R4-type instructions possible to implement. P extension would be able to also put own ternaries in remaining space.

Vector ext dropped those encodings for its VFMA4 due to much more serious issues than arising from regular scalar ternaries.

In this way OP space won't get polluted and there will be very little overhead on zfinx (or P) enabled microarch as those already have to implement 3rd read port or multi cycle execution. Also, an acces to 3rd port (or second cycle exec) could be determined by just inst[6:4] according to table 24.1 in base spec document.

jnk0le, 2021-03-31:
@clairexen 
I think that might be better than c3/c4 formats overall. Only not that much space.

clairexen, 2021-04-01:
Where would you put the immediate versions of the shift-like ternary instructions? ideally each one of those should provide up-to 7 immediate bits to accommodate RV128, i.e. 4x the size of a ternary instruction getting it's 3rd argument from a register. the proposal I shared on the list defines 4 such instructions FSLI, FSRI, SAPI, CUTI. If we would put them there as well, then they alone would already take up half of that encoding space. If we put them somewhere else then your proposed solution only provides about one quarter of the encoding space we'd need for FSL, FSR, SAP, CUT, MIX, MUX, FSLI, FSRI, SAPI, CUTI.

Edit: Obviously FSLI and FSRI are redundant. facepalm I'll update my proposal to only contain FSRI..

clairexen, 2021-04-01:
I do however like the idea of using the reserved FMA encoding space a lot. It would make decoding a bit easier than what I had proposed. However, I would dedicate one major opcode each to the R4, C4, C3, and C2 formats.

Without use of the immediate bits that C2/C3/C4 provide, that would create enough encoding space for 8 R4 instruction, 2048 C4 instructions, 16384 C3 instructions, and 131072 (unary) C2 instructions. Or with all immediate bits used, 32 of each of the C2/C3/C4 instruction formats:

```
c2.0          64 cp
c3.0         512 cp (  0.5 kcp)
c4.0        4096 cp (  4.0 kcp)
-----
c2.12     262144 cp (256.0 kcp)
c3.9      262144 cp (256.0 kcp)
c4.6      262144 cp (256.0 kcp)
-----
r4       1048576 cp (  1.0 Mcp)

c4.0 in reserved FMA space: 8192
c4.6 in reserved FMA space: 128
r4   in reserved FMA space: 32

c2.0  in reserved FMA space in one opcode: 131072
c3.0  in reserved FMA space in one opcode: 16384
c4.0  in reserved FMA space in one opcode: 2048
-----
c2.12 in reserved FMA space in one opcode: 32
c3.9  in reserved FMA space in one opcode: 32
c4.6  in reserved FMA space in one opcode: 32
-----
r4    in reserved FMA space in one opcode: 8
```

(cp = codepoint, kcp = 1024 cp, Mcp = 1024 kcp)

jnk0le, 2021-05-18:
Using all those encodings would conflict with potential posit addition.
Probably we should use only one rm per opcode if we care about it.

https://arxiv.org/pdf/1908.01466.pdf