riscv/riscv-bitmanip #115: Test failed on master branch

mohanson, 2021-02-18:
Well, when I noticed that v0.93 was released sometime in the past, I tried to check out the latest code, rebuilt the tool, ran the test, and then this problem occurred 

```sh
$ cd riscv-bitmanip/tests
$ sh run.sh 

+ gcc -o testgen -Wall -O1 -I../cproofs testgen.cc
+ ./testgen
+ /opt/riscv64b/bin/riscv64-unknown-elf-gcc -Wall -march=rv64gb -Os -o tests tests.c
+ /opt/riscv64b/bin/spike --isa=RV64GCB pk tests
bbl loader
test_zbb  0x6ee99663 ERROR
```

My environment is as belows, they are all built by `riscv-bitmanip/tools/build-all.sh`

```
riscv-bitmanip
    commit: 2d5ca17db360f1083ab7b3867071518ae442208b(latest)

riscv64-unknown-elf-gcc
    --version: riscv64-unknown-elf-gcc (GCC) 10.0.0 20190929 (experimental)

spike
    --version: Spike RISC-V ISA Simulator 1.0.1-dev, 
    commit: 8faa928819fb551325e76b463fc0c978e22f5be3(latest)
```

I can't determine where the problem is :)

mohanson, 2021-02-24:
I think gcc mistaken the encoding of `max` and `minu` when compiling, after modifying the code in https://github.com/riscv/riscv-bitmanip/blob/master/tests/test_zbb.c#L76-L80 `test_zbb` can run normally,

But at the same time spike will hang when executing `test_zbs`, I suspect that the gcc and spkie versions are inconsistent.

```text
+ /opt/riscv64b/bin/spike --isa=RV64GCB pk tests
bbl loader
test_zbb  0x90f45e4e OK
test_zbs <-----------------hang
```

rdolbeau, 2021-02-24:
@mohanson Yes the current binutils have a bug, see #93

jim-wilson, 2021-02-24:
That binutils bug was fixed in December.  Maybe you are using the wrong commitid or wrong branch?  I don't know how riscv-bitmanip testing works.

mohanson, 2021-02-24:
It can be easily reproduced, as long as you run the following command 

```
$ git clone https://github.com/riscv/riscv-bitmanip
$ cd riscv-bitmanip

$ cd tools
$ sh build-all.sh

$ cd ..

$ cd tests
$ sh run.sh
```

jim-wilson, 2021-02-24:
It is probably riscv-bitmanip that is out of date.  It hasn't been maintained properly for months.  The main maintainer hasn't had enough time to work on it.