riscv/riscv-bitmanip #120: zext.w missing from zbb table

jim-wilson, 2021-02-25:
The table at the beginning of chapter 2 which lists all of the instructions in each subset doesn't mention zext.w for zbb.  It is supposed to be in there.  It is an alias of add.uw which is zba, but the subset that provides zext.w is in zbb.  Just like zext.h in zbb is an alias for pack/packw even though pack/packw are not in zbb.

The idea here is that the base architecture plus zbb is supposed to provide a complete set of [sz]ext.[bhw] instructions.  This set is not complete unless we have zext.w for rv64.

kdockser, 2021-02-25:
> CAUTION: This email originated from outside of the organization.

Hi Jim,

The instruction zext.w was removed from Zbb on purpose. Perhaps this was because it had originally been defined as an alias for pack rd, rs1, zero which had been removed from zbb and it wasn’t considered worth the (small) cost.  However, you are correct that this could also be an alias for addu.w rd, rs1, zero. We could add it back as an alias into zba – not quite what you are looking for.

That said, pack is needed for crypto. It might make sense to add pack back into Zbb and then we can get zext.w for free again.

Ken

jim-wilson, 2021-02-26:
zext.w is already mentioned in multiple places in the document as an alias for add.wu.  I thought we had agreement that the zext.w subset of add.wu would be part of zbb, but Andrew Waterman tells me it isn't.  Oh well.  This means that I plus zbb has 5 of the 6 possible [zs]ext.[bhw] instructions, and you need I plus zbb+zba to get all 6 of them.  But anyone sensible should be implementing zbb+zba anyways, so we can live with that.  In the short term, we have a few broken toolchains floating around, because the toolchain folks thought that zext.w was in zbb, but this will be a temporary problem.

kito-cheng, 2021-03-02:
Oops, the current implementation in gcc is generating `zext.w` for zbb or zba, seems like it should fix...

RogierBrussee, 2021-03-03:
Maybe slli.uw for imm = 0  instruction should be the alias for zext.w  and in zbb.
IIUC it is preferred to special case an immediate over special casing rs1 or rs2 == zero. 

Or maybe it is not too late to include the whole slli.uw instruction in Zbb, so that with just I + Zbb you can do sensible unsigned index -> address computation even if you don't want to implement the fused slli.uw + add 
of Zba (but as Jim rightly remarks, implementing Zba is the sensible thing to do).