riscv/riscv-bitmanip #125: Bit 31 is always zero in these 64-bit test values (non-uniform random). Also - must these be nonnegative?

mjosaarinen, 2021-04-15:
https://github.com/riscv/riscv-bitmanip/blob/64ff098dc0aec264b1c4cf359b30a67ec03bb718/asmtests/auto/asmgen.c#L52

Bit 31 is always zero in these 64-bit test values. Also - must these be nonnegative?

eroom1966, 2021-04-15:
@mjosaarinen 
> Bit 31 is always zero in these 64-bit test values
well spotted!
it appears that RAND_MAX is (2**31 - 1), therefore this function (random()) returns a number between 0x00000000 0x7FFFFFFF
could you look at creating a fix and issuing a PR ?

> Also - must these be nonnegative?
I don't understand the question, this function should generate a value between 0 and (2**XLEN - 1)

thx
Lee

mjosaarinen, 2021-04-15:
It creates a number between 0 and 2**(XLEN-1)-1. The sign bit is also always zero.

I'm not sure if I know how to test the testing thing as I've never used it, but sure I can try.. The tests will be different, how many things that will break?

mjosaarinen, 2021-04-16:
Well, it is a trivial bug to fix but unfortunately, I don't have the tooling available at the moment to test that everything works after the fix; so I suggest that someone else does it. Note that the missing bits at critical locations of stimulus words significantly lessen the coverage of this particular testing suite, so I'd suggest giving this some priority.