riscv/riscv-bitmanip #128: Comments on version 94.60 (pdf)

PardoIlan, 2021-05-19:
1. The encoding of max and minu have been flipped (changed between v0.92 and 0.93) - Is it intentional? Seems there was a change in the instructions name order in the table which was not done on the encoding.
2. The encoding of rori for RV64 bit mode is different from that in RV32. It was the same in 0.93 and before - Is this change intentional? 
3. The source operand which get zero extended was changed from rs2 in v0.92 to rs1 (consistent with as slli.uw) between v0.92 and v0.93 - Is this intentional? I guess yes.

Thank you

jim-wilson, 2021-05-20:
1 and 3 are deliberate, and were discussed on the mailing list.  These were to make the instruction encodings more consistent with other extensions, and hence simpler to decode.

2 I hadn't seen before and looks odd.

PardoIlan, 2021-05-20:
Thanks for the prompt response

ptomsich, 2021-06-22:
2 is similar to what has been also discussed in #146 : this change makes the size of the shamt-field and the undefined-instruction handling on RV32 explicit (i.e. only 5 bits of shamt can be encoded).

closed 2021-06-22 by ptomsich