riscv/riscv-bitmanip #143: rol, ror and rori have a similar bug in their Sail pseudo-code

slobodova, 2021-06-12:
Chapter 2, Sail code for ror, rol and rori uses X(rs2) instead of X(rs1) in the shift. X(rs2) should be used only for the computation of shamt.
One more comment, I assume that when X(rd) is used on the left-hand-side  and right-hand-side has more than XLEN bits, the low XLEN bits are taken. But I haven't seen any comment about it.

rol current:
let shamt = if xlen == 32
  then X(rs2)[4..0]
  else X(rs2)[5..0];
let result = (X(rs1) << shamt) | (X(rs2) >> (xlen - shamt));
X(rd) = result;

rol suggested:
let shamt = if xlen == 32
  then X(rs2)[4..0]
  else X(rs2)[5..0];
let result = (X(rs1) << shamt) | (X(rs1) >> (xlen - shamt));
X(rd) = result;
---------------------------------
ror current:
let shamt = if xlen == 32
  then X(rs2)[4..0]
  else X(rs2)[5..0];
let result = (X(rs1) >> shamt) | (X(rs2) << (xlen - shamt));
X(rd) = result;

ror suggested:
let shamt = if xlen == 32
  then X(rs2)[4..0]
  else X(rs2)[5..0];
let result = (X(rs1) >> shamt) | (X(rs1) << (xlen - shamt));
X(rd) = result;
--------------------------------------------
rori current:
let shamt = if xlen == 32
  then shamt[4..0]
  else shamt[5..0];
let result = (X(rs1) >> shamt) | (X(rs2) << (xlen - shamt));
X(rd) = result;

rori suggested:
let shamt = if xlen == 32
  then shamt[4..0]
  else shamt[5..0];
let result = (X(rs1) >> shamt) | (X(rs1) << (xlen - shamt));
X(rd) = result;

ptomsich, 2021-06-12:
Thanks for the reviews of the SAIL code, very much appreciated; every helping hand on SAIL is currently needed.
Could you submit pull-requests for these changes?  This would allow us to queue them up already and avoid any mistakes in applying your suggestions ourselves.

slobodova, 2021-06-12:
Do you have any special procedure for that?

closed 2021-06-21 by ptomsich

M-Ijaz-10x, 2022-03-14:
The issue with the syntax of rol, ror, and rori is still not fixed in the bitmanip-1.0.0 public review document. Is this intentional or you are waiting on the next release. cc@cetola

ptomsich, 2022-03-14:
This is fixed in one of the follow-on PDFs (1.0.0-<num>.pdf)

However, I have been bringing up the need to create a final document repeatedly at Tech Chairs without any result.
I'll try to escalate this topic once more.

M-Ijaz-10x, 2022-03-14:
@ptomsich thanks