riscv/riscv-bitmanip #147: Should slli.uw have opcode OP-IMM instead of OP-IMM-32?

JamesKenneyImperas, 2021-06-17:
The current encoding for `slli.uw` with opcode `OP-IMM-32` seems inconsistent to me. In the base architecture, this normally means that the encoded immediate (if a shift) is constrained to the range 0...31 (see section 5.2 in the Unprivileged ISA Specification). I think that here values 0...63 are allowed, in which case I would expect it to have opcode `OP-IMM`.

JamesKenneyImperas, 2021-06-17:
Thinking about this more, I wonder whether the encoding should stay where it is but the shift amount should be constrained to 5 bits (0...31). Shifts in the range 32...63 are equivalent to `slli`.

ptomsich, 2021-06-21:
While this doesn't contribute directly to resolving this, it is worth noting that the other `.uw`-instructions are OP-32.

aswaterman, 2021-06-21:
It's six of one, half a dozen of the other.  We need the OP-32 control signal to determine that we need to zap the MSBs.  OTOH, we want the OP-XLEN control signal to write the full 64b result.  So, moving it would fix one problem and create another.  But, either way, we're talking about ~1 gate.

(I'll record this in the public review feedback.)

closed 2021-06-21 by aswaterman

JamesKenneyImperas, 2021-06-22:
Hello Andrew,

Just to clarify what you are thinking: would you expect a shift in the range 32...63 to be an Illegal Instruction, or not? To me, it would seem more consistent with the base architecture if it _was_ illegal, and there seems little point allowing these shifts since they are equivalent to `slli` anyway and hence redundant.

Earlier versions of the B extension specified shifts for `GORCI`/`GREVI `in an inconsistent way compared to the base, so I'm glad to see that this seems to have been fixed in this 1.0.0 version. `slli.uw` is the one remaining potential outlier in the up-for-ratification subsets I think.

James.

aswaterman, 2021-06-22:
It's true that, for shift amounts >= 32, `slli.uw` is redundant with `slli`.  It's basically harmless, though (no additional datapath cost).  And the RV128 version of `slli.uw` will need all 7 bits, because only immediates 96-127 will be redundant with `slli`.  For simplicity in that case, we would definitely just allow all patterns 0-127, so for consistency back in RV64, allowing patterns 0-63 is consistent.

JamesKenneyImperas, 2021-06-23:
Ok, thanks. It seems a little odd to me that `slli.uw` allows shifts 0...63 when `slliw` is limited to 0...31, but as long as it is clear in the specification how the shamt is treated then this is fine.