riscv/riscv-bitmanip #151: Is GREV gone

Serentty, 2021-07-02:
Sorry for opening an issue for this—this reposistory doesn't have discussions enabled, and I haven't been able to find any discussion about this on the web. Was the GREV instruction removed from the B extension? I can't find it anywhere in the latest proposal.

JamesKenneyImperas, 2021-07-02:
GREV is not a member of the subsets being submitted for public review (Zba, Zbb, Zbc and Zbs) so not described in this document. I think GREV is in the Zbp subset.

Serentty, 2021-07-02:
Oh thanks! Where can I find the Zbp specification?

JamesKenneyImperas, 2021-07-02:
I don't know the answer to that one. @ptomsich, can you help?

ptomsich, 2021-07-02:
I haven't seen any update on this (as part of the info I have on this year's ISA extension priorities) and can only confirm that it is also not part of the Zbk[bcx] package (which is in-flight together with scalar crypto).

I don't know if there is a consensus around the content of Zbp at this time.

jim-wilson, 2021-07-02:
See the docs on the main-history branch.  Looks like there was a major change in May that replaced the old riscv-bitmanip tree with a new one, and the new one only has info on parts that are being submitted to a vote.  But at least the old info is still available on a branch.

mlawson-tt, 2023-04-07:
It's been a while here, but does anyone have any updates on the Zbp extension (i.e. how likely it is to be ratified and any potential timelines)?  We're debating whether we should use `grevi` or `rev8` nomenclature, and `rev8` is in a ratified extension, but `grevi` is a superset of `rev8`.  If Zpb is likely to be ratified (or even just any extension that makes `rev8` a pseudo-instruction for `grevi`), we'd likely go with `grevi` but otherwise we'd use `rev8`.  Same goes for `orc.b` vs `gorci` (and potentially others).

kdockser, 2023-04-07:
As is the policy of RVI, the BitManip task group was dissolved once the original BitManip Extensions (Zva, Zvb, Zvc, and Zvs) were ratified. If there is enough interest, someone can request to spin up a new Task Group to standardize additional vector or scalar BitManip instructions.
What is the context for your debating whether to use `grevi` or `rev8` nomenclature? Is this for a SIG, TG, software infrastructure, or some other area?

mlawson-tt, 2023-04-07:
It's just for a software simulator that logs disassembled instructions as they are executed.  `grevi` is already implemented, but the desire is to have `rev8` logged in the disassembly.  As it stands, dynamically printing `rev8` instead of `grevi` when Zbb is enabled but Zbp is not would require a large number of changes (that may end up being necessary eventually), but if `grevi` (and others) weren't going to be standardized anytime soon, we'd just simplify the `grevi` implementation and convert it to `rev8`.

sh1boot, 2023-08-20:
Well I've come sniffing around here not because I'm looking for the full `grevi` but just the basic bitwise bit reversal.

I've used this pretty heavily in bitstream parsing on various architectures and hoped to carry on doing so on RISCV, but the only trace I can find of it is in Zvbb, for vector.

So mark this random passer-by down as "interested", for whatever that's worth.