riscv/riscv-bitmanip #167: GCC toolchain for draft v.0.93 (and v.1.0.0)

vogelpi, 2021-12-07:
Upstream GCC has now support for bitmanip v.1.0.0. But AFAIK there is no toolchain available that supports v.0.93 containing both the instructions in v.1.0.0 plus quite a few additional instructions that haven't been ratified at this point. Nevertheless, there are hardware implementations that also support the non-ratified parts of v.0.93 such as our [Ibex core](https://github.com/lowrisc/ibex). In order to verify this implementation, I've come up with a binutils version that supports v.0.93 and I think sharing this work would potentially be useful for others as well.

Is there interest from the bitmanip working group that I prepare a PR to update [the version referenced on the main-history branch under tools](https://github.com/riscv/riscv-bitmanip/tree/main-history/tools)? This one is currently on draft v.0.92 containing quite some changes compared to draft v.0.93.

Imperas, 2021-12-07:
@vogelpi  - I personally think this would be a good idea - at Imperas our simulators can be configured to select any of the pdf'd versions - and we always have customers asking for specific toolchains for the version they have in silicon/rtl - we keep some of these compiled up in our public github in different branches https://github.com/Imperas/riscv-toolchains/branches/all - and have bitmanip 0.92 and 1.0.o - if you do a 0.93 we could use it. Thanks. Simon.

kito-cheng, 2021-12-07:
GNU toolchain has policy that is only accept ratified extension, so according the policy we won't upstream non-ratified ext. to upstream GCC/binutils.

**BUT** we (RISC-V GNU toolchain(GCC/binutils) maintainers) has a discussion about the upstream policy recently, we are consider to change the policy to follow RISC-V LLVM policy - accept non-ratified ext. with `-menable-experimental-extension` scheme, so ideally you can upstream those non-ratified exts. in future.

However GCC development is under stage 3 which is not allow new feature, so we think it's not good timing for doing policy changing, the time after GCC 12 release would be a better, which might be April, 2022.