riscv/riscv-bitmanip #169: Question: No bit-field extract/insert instructions?

mbitsnbites, 2022-01-03:
What is the rationale for not including proper bit-field manipulation instructions in the bitmanip extension?

Besides being very useful for extracting / inserting bit-fields from / to integer registers, they can also be used for:

* Sign- and zero-extension of any size type (i.e. SEXT/ZEXT become redundant).
* Masking out lower N bits (i.e. implement `ANDI rd, rs, (2^N-1)`).
* Arithmetic and logic left/right shift (i.e. SLL/SRL/SRA become redundant, though that ship has already sailed since they are part of the base ISA).

jim-wilson, 2022-01-04:
Many instructions were proposed in the bitmanip spec, and only a subset of them made them through the approval process.  The full proposal did include some bitfield extract and insert instructions.  You can find this info on the main-history branch.  See bcompress/bdecompress in later versions, or bext/bdep in earlier versions.  I don't know the exact reasons why bitfield extract/insert didn't make it into the final spec.  Maybe someone thought they weren't useful enough.  Maybe someone thought that they were too hard to implement in hardware.  Maybe the compiler patches weren't good enough to show how useful they were.  Maybe someone thought that the bitmanip spec was using too much of the opcode encoding space.  Or maybe something else.  It is possible that more instructions from the bitmanip spec will be added in future ISA updates.  With about 16 ISA extensions approved last year, there were simply too many extensions in flight to get everything through the approval process.
[

mbitsnbites, 2022-01-04:
I get that.

I was just surprised to find out that bit-field instructions had been dropped, especially when reading this in the latest RISC-V spec (chapter 18, "B" extension):

> This chapter is a placeholder for a future standard extension to provide bit manipulation instructions, **including instructions to insert, extract, and test bit fields**, and for rotations, funnel shifts,
and bit and byte permutations

I get the feeling that the Zbs extension (single-bit instructions) is the reduced functionality version of full bit-field instructions (i.e. bit-field instructions with a fixed `width = 1` specifier). Looking at the instruction encoding, I also get the feeling that it simply was not possible to include the width specifier (5-6 bits) in the 32-bit instruction word.

Last year I discovered how versatile the bit-field instructions are (the M88K-style ones), so I went ahead and [replaced ASR/LSR/LSL in my MRISC32 ISA with EBF/EBFU/MKBF](https://www.bitsnbites.eu/mrisc32-stabilizing-the-base-architecture/) (I also added IBF - Insert BitField). They are really basic building blocks of an ISA (as they can replace several other instructions).

As it turns out that most ISA:s end up including bit-field instructions (ARMv7, AArch64, POWER, x86 (SIMD), etc, all have them to some extent), I figure that the sooner you add them to the ISA, the better (otherwise you will end up with lots of overlapping functionality that wastes opcode space).

Daydream0929, 2025-11-03:
Is there any further progress regarding the bit-field instructions?